smartin015/continuousprint #3: Beta releases / Changelog

smartin015, 2022-02-07:
I'll ping this thread whenever I push a new version to master. Doing this instead of a regular release so as to keep versions in sync with the versions at https://github.com/Zinc-OS/continuousprint. If you see any bugs, please create a new issue in this fork rather than commenting on the release. Thanks!

In the latest merge, I've added my changes from [prints_are_files](https://github.com/smartin015/continuousprint/tree/prints_are_files) and from [sortable_jobs](https://github.com/smartin015/continuousprint/tree/sortable_jobs). This allows for

* use of `@pause` and other at-commands in the settings of the continuous print queue
* collecting individual queue items into "jobs" which can be run many times (standing in as a "loop" function)
* Draggable reordering of items within jobs, between jobs, and reordering of jobs themselves
* A "requeue failures" option that allows for failed prints to be automatically marked for re-printing

[image]

TylonHH, 2022-02-07:
How do I install your fork on OctoPrint? 😄

piebaker, 2022-02-07:
Install plugin via URL in octoprint:
https://github.com/smartin015/continuousprint/archive/master.zip

TylonHH, 2022-02-08:
ok I got it, but I cant add any gcodes. I see the + (plus) "add to CPQ" but nothing happens.

TylonHH, 2022-02-16:
> > `Could not bind view model CPViewModel to target #tab_plugin_continuousprint : visible@http://192.168.178.68/static/webassets/packed_libs.js?1fe9f29c line 131 > Function:3:60`
> 
> @TylonHH Looks like a frontend error, but that's all I can get out of that error :( can you link to a [paste](http://pastebin.com) of the whole console output? Maybe there's something more there...

https://pastebin.com/9S6YMpuz here you are. Is this what you need?

smartin015, 2022-02-18:
Ugh, sorry for the stale-bot. Apparently it snuck in when I originally forked - the config was in a hidden folder `.github/workflows/stale.yml`. 

@TylonHH sadly that paste doesn't show any useful details either, it's all just packed octoprint code, nothing to do with the continuous print plugin. If there are any logs *before* that error there might be something useful there. Otherwise, try copying your plugin settings (gcode etc) and clicking the "Cleanup Plugin Data" eraser icon in the plugin manager next to the CPrint plugin, then restarting octoprint. 

Thanks for bearing with me on this - I'm not sure what's unique about your setup but I'd like to get it fixed.

smartin015, 2022-03-13:
Now that ownership has changed over, these changes are available as [v1.4.1](https://github.com/smartin015/continuousprint/releases/tag/1.4.1) from the plugin repository.

I'm going to close this out. @TylonHH, please open another issue if you keep seeing errors - thanks!

closed 2022-03-13 by smartin015