smartin015/continuousprint #10: Document API

smartin015, 2022-03-13:
Partner issue to https://github.com/Zinc-OS/continuousprint/issues/58. Currently integration isn't well documented, and checked-in examples would be useful.

smartin015, 2022-03-17:
@aburtonProto - I'm working to "officially" declare an API for the continuous print plugin, and I'd appreciate your input.

Can you take a look at the [new API documentation](https://github.com/smartin015/continuousprint/tree/rc#integrate-with-scripts--other-plugins--other-devices) in the `rc` branch and let me know your thoughts? 

I looked briefly at https://github.com/aburtonProto/OctoPrint-Protosthetics and it seems like you only use `set_active`, so this should be a no-op on your end. But if you want, you can also install from `https://github.com/smartin015/continuousprint/archive/refs/heads/rc.zip`, which currently has some changes to the unlisted handlers that returns a "deprecated" notice. 

Thanks!

aburtonProto, 2022-03-17:
That is correct, we are only using the set_active right now.  
I appreciate that in the api_examples folder you have one for both python and javaScript.
For our use case, we are managing printers remotely.  We queue up the next prints for the clients so when the print is finished they clear the bed and hit a big button to start the next print.  For this, set_active and the @pause command within the bed clearing script are what we use from the API.  
I do also set_active to False when the printer disconnects, as we had some amusing things happen otherwise.

I installed 1.4.2rc1 today and am running that.  So far all is well.

smartin015, 2022-03-18:
Great to hear this is a compatible change! As time permits, I'll do a release and let the deprecation soak a bit so as to avoid catching anyone else off-guard.

> managing printers remotely

You might be interested in #35 - I'm very early into thinking about support for "one queue, many printers". Oh, and also #34 which would allow for "many users, one queue". Combining the two could open up some interesting possibilities... I'd welcome some early input on those issues if either are useful to you :)

aburtonProto, 2022-03-18:
#35 has my attention more than #34, but the combination I could see as
quite powerful.
I am excited by all the planned enhancements you have, and will gladly help
with testing where I can.
As you mentioned that you looked at my plugin, you know my plugin skills
are still developing (pun intended).
If you would like to skip going through gitHub for communication, my email
is ***@***.***
If you would rather keep gitHub as the middle man, I am ok with that as
well.

smartin015, 2022-03-19:
> #35 has my attention more than #34, but the combination I could see as quite powerful. I am excited by all the planned enhancements you have, and will gladly help with testing where I can. As you mentioned that you looked at my plugin, you know my plugin skills are still developing (pun intended). 

Great - I absolutely need early testers to help make releases more stable. With #22 we now have a `Release Candidate` branch which you can opt-in to get all pre-release updates (generic instructions [here](https://community.octoprint.org/t/how-to-use-the-release-channels-to-help-test-release-candidates/402)).

Can you turn this on for your setup?

> If you would like to skip going through gitHub for communication, my email is ***@***.*** If you would rather keep gitHub as the middle man, I am ok with that as well.

Let's stick to GitHub for now - it's easier for me to have everything in one place :)

closed 2022-04-11 by smartin015