smartin015/continuousprint #6: Wait for bed cooldown not consistent between printers.

smartin015, 2022-03-13:
Partner issue to https://github.com/Zinc-OS/continuousprint/issues/69

willindiana, 2022-03-14:
The latest changes are located in this fork https://github.com/willindiana/continuousprint

Will be reconciling this feature with latest changes here. Pull Request is to come in the next few days.

smartin015, 2022-03-15:
Hey @willindiana - I took a brief look at your changes, and it looks like you might be calling `time.sleep()` inside a handler which gets invoked from the event loop. According to octoprint this can block events / "break the server" ([link](https://docs.octoprint.org/en/master/plugins/mixins.html?highlight=long%20running#eventhandlerplugin)). 

When you adapt it to the changes in the new repo, I'd suggest looking into [`octoprint.util.RepeatedTimer`](https://docs.octoprint.org/en/master/modules/util.html?highlight=threading#octoprint.util.RepeatedTimer) or `ResettableTimer`, which seems to handle the waiting action asynchronously in a separate thread.

willindiana, 2022-03-24:
Quick Update, I have ported my changes to this branch last weekend, and removed all `time.sleep()` instances from my implementation. Instead, I followed @smartin015's advice and used an implementation of `octoprint.util.RepeatedTimer`. I will be testing this weekend and making all necessary changes.

DanaViolet, 2022-03-29:
Thank you both for working together on this issue.  😎👍
Once it is implemented I'll be updating to the new version.

willindiana, 2022-04-02:
Hey @smartin015 got pretty busy with work but I finished and have tested my changes for this bug Here's the pull request looking forward to your review!
https://github.com/smartin015/continuousprint/pull/44

FYI. I ended up simplifying my approach I still removed all `time.sleep`'s but I also decided not to use util.RepeatedTimer. I ended up going with my original approach and using a timeout-controlled while loop.

smartin015, 2022-04-11:
@willindiana, @DanaViolet the change is now merged in and I've tagged it as release `1.5.0rc3`. 

Can you both give it a test drive by switching to the "Release Candidate" channel under Settings > Software Update?

[image]

If everything's working as expected, I'll promote it to a full release. Thanks!

smartin015, 2022-04-15:
@willindiana, @DanaViolet any luck?

willindiana, 2022-04-15:
@smartin015 I haven't had a chance quite yet but I'll be running some prints today and tomorrow! I'll put my test results here :)

willindiana, 2022-04-16:
@smartin015 Update I have been running the plugin most of the day yesterday and so far today. Smoke tests all passed and both timeout and threshold are working as expected.

closed 2022-04-17 by smartin015

smartin015, 2022-04-17:
It's now published under `v1.5.0`

PR #51 (2022-04-17): Merge v1.5.0
Upstreams CI testing, new documentation, and bed cooldown functionality.

Closes #6