smartin015/continuousprint #8: Chromium virtual keyboard compatibility

smartin015, 2022-03-13:
Partner issue to https://github.com/Zinc-OS/continuousprint/issues/64

merijnpeters, 2022-03-14:
Tested with the new version (1.4.1) but it still doesn't work. I'm also missing the "File list" now

smartin015, 2022-03-15:
Thanks for confirming. The file list was actually removed, as it was redundant to the list on the left panel (you can click the `+` button there now to add to queue)

Just to confirm, [this](https://chrome.google.com/webstore/detail/virtual-keyboard/pflmllfnnabikmfkkaddkoolinlfninn?hl=en) is the extension you're using? If so, I can confirm it doesn't appear when clicking on one of the inputs in the plugin. I'll do some more digging to figure out what's the problem.

smartin015, 2022-03-15:
Ok, I think I found and fixed the problem. Can you follow the `.zip` instructions at https://github.com/smartin015/continuousprint#add-the-plugin to install from the master branch and confirm?

smartin015, 2022-03-20:
Hey @merijnpeters - since the keyboard extension works on my setup, I'm going to go ahead and close this out. Please reopen the issue if you continue to see problems. Thanks!

closed 2022-03-20 by smartin015