smartin015/continuousprint #25: when I start the queue with other files, it starts the clean bed script then start the printing process

sidneyns, 2022-03-16:
**Describe the bug**
When I start the script, it start the bed cleaning script then starts the printing process. 
I have added a new cleaning bed script , however it shouldnt make any diference. 

What I also noticed is that when I place the file on the queue, If I only have one file it works fine ( start the heating process of the file). It usually gets mess up when I have other files that has been already processed it in the past.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
It should start the printing process since its the 1st print of the day

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System info:**
Follow the instructions [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) to generate an OctoPrint system bundle, and attach it to this issue.

**Browser logs:**
Collect the console logs for your browser and upload them to this issue. See [here](https://support.shortpoint.com/support/solutions/articles/1000222881-save-browser-console-file) for instructions on how to do this for different browsers.

**Additional context**
Add any other context about the problem here.

smartin015, 2022-03-17:
Thanks for the report @sidneyns - let's investigate a bit more to find the problem. Please follow these steps:

1. Add multiple prints to your queue (in a way which will trigger the bug)
2. Start the queue
3. Let it run until it's started clearing the bed (wrongly)
4. Follow instructions [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) to download a system bundle file
5. Save the console output from the octoprint page (instructions [here](https://support.shortpoint.com/support/solutions/articles/1000222881-save-browser-console-file))
6. Upload both of these files to a comment on this issue

This will help me diagnose what's going wrong, and hopefully fix it. Thanks!

smartin015, 2022-03-23:
Hey @sidneyns - any chance you were able to generate a system bundle and console output so I can troubleshoot further?

sidneyns, 2022-03-24:
Hello.. I am testing again your version and so far so good...
you can close the ticket

closed 2022-03-24 by smartin015