smartin015/continuousprint #29: Queue only starts the first print and then is unable to continue

Tiri1i1, 2022-03-17:
**Describe the bug**
When adding several prints to the queue, only the first print works and then the rest does not get printed (the plugin keeps managing the queue but does not launch prints again)

**To Reproduce**
Steps to reproduce the behavior:
Have a CR-30 Printer with Klipper set up
Use this G-Code as bed cleaning script: 
G28 X0 Y0 Z0
G91 ; Set relative for lift
G0 Z10 F200
G90 ;back to absolute positioning
G28 X0 Y0 Z0; home
Add 2 or more prints to the queue
Launch the management of the queue
See that only the first part is printed

**Expected behavior**
All parts in the queue should be printed

**System info:**
CR30 With Klipper
[octoprint-systeminfo-20220317155654.zip](https://github.com/smartin015/continuousprint/files/8287861/octoprint-systeminfo-20220317155654.zip)

**Browser logs:**
Both tested on Firefox and Chrome, no difference in behaviour.

fixing-it, 2022-03-18:
same here with my Prusa MK3s 
rarely it starts a second print, but couldn't figure out why. 3 instances of octoprint and started the same queued files on all 3 printers.

Tiri1i1, 2022-03-19:
Update: Still the same behaviour after updating to 1.4.2

smartin015, 2022-03-19:
Thanks for the report - I took a look at your `octoprint.log` and I found:

```
2022-03-17 09:20:30,286 - octoprint.plugins.klipper - ERROR -  Must home axis first: 0.000 50.000 74.788 [53579.479]

2022-03-17 09:20:30,289 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: Must home axis first: 0.000 50.000 74.788 [53579.479]
```

It seems to be erroring when it encounters your `G0 Z10 F200` command. Interestingly, there's no indication it actually sent the first line in the logs:

```
| Send: M105
| Recv: ok B:62.3 /0.0 T0:104.1 /0.0
| Send: G91
| Recv: ok
| Send: M105
| Recv: ok B:62.0 /0.0 T0:101.1 /0.0
| Send: G0 Z10 F200
| Recv: !! Must home axis first: 0.000 50.000 74.788 [53579.479]
```
Here's a couple things to try:

1. Add a duplicate `G28 X0 Y0 Z0` command at the top (if this works, it's a parsing bug on the plugin side)
2. Strip off `X0 Y0 Z0` from the `G28` command (maybe it thinks there's another axis not homed?)
3. If you use a BL-Touch, switch to "safe z-homing" (described [here](https://www.klipper3d.org/BLTouch.html), suggested [here](https://github.com/Klipper3d/klipper/issues/2272))
4. Check for correct [homing override](https://github.com/Klipper3d/klipper/issues/2717)

Other issues [here](https://github.com/Klipper3d/klipper/issues?q=is%3Aissue+%22Must+home+axis+first%3A%22) might be informative. I don't use Klipper myself, but it seems like the issue is Klipper raising an alarm because it doesn't think it's been homed.

Let me know whether any of these help!

fixing-it, 2022-03-19:
I don’t use Klipper and have the same problem. Stock Prusa Mk3s.
The last zinc version 1.31 of the plug-in works for me

smartin015, 2022-03-20:
Hey @fixing-it - didn't realize you weren't also running Klipper. Since your setup is different, please file a separate [bug report](https://github.com/smartin015/continuousprint/issues/new?assignees=&labels=bug&template=bug_report.md&title=) and follow the system info & browser logs steps so I can investigate. Thanks!

smartin015, 2022-03-23:
Hey @Tiri1i1, have you tried any of the changes I suggested? How'd it go? :)

smartin015, 2022-04-03:
I'm going to go ahead and close out this issue as it's been a couple weeks and I can only assume you found a fix. Please let me know if you're still having trouble with Klipper + CP and I can reopen it.

closed 2022-04-03 by smartin015