smartin015/continuousprint #34: Multi-user support

smartin015, 2022-03-17:
**Is your feature request related to a problem? Please describe.**

The queue has no notion of which OctoPrint user added which job item. This is a hindrance from automating prints in an environment where many users are printing (e.g. a large club, campus, or business)

**Describe the solution you'd like**

Multi-user queue modification with attribution. It should:

* Indicate which user owns which print jobs
* Disallow modifying others' jobs
* Have some reasonable (configurable?) behavior when one user tries to preempt another user

smartin015, 2022-06-02:
With the refactor in `v2.0.0rc1` this should now be much easier - I think it needs:

* [ ] A `creator` column for jobs that stores the user which created it
* [ ] An added "queue admin" Permission which allows mutating jobs which the user does not own
* [ ] Checks in API calls that `creator` matches the user making the call
* [ ] UI changes to disable features for which the logged-in user has insufficient permissions

smartin015, 2022-06-08:
Recently published: https://github.com/chennes/OctoPrint-Queue - a queuing system designed for library use.

chennes, 2022-06-08:
ContinuousPrint is substantially more sophisticated than Octoprint-Queue -- OQ allows basically totally arbitrary text is each of the fields, and isn't tied to who is logged into Octoprint.