smartin015/continuousprint #39: cleaning script starts before set temperature is reached

sheldo26, 2022-03-20:
since upgrading to version Continuous Print (1.4.2), i have been getting the following error

when completing the "job" the cleaning script is to wait until the bed temp is at 27 degree before starting the cleaning movement.
(between 30 - 27 degree the prints release from the bed)

the issue is that the cleaning script is not waiting for the request bed temp before "cleaning" the bed, 3 times i have attempted using version 1.4.2. each time the cleaning script is has started running when the bed still about the mid 40 degrees
when means the print has not release from the bed, when the hotend goes to clean the bed.

on the my ender 3pro, at end of any print sets the hotend to y235 x117.5

my cleaning script, 

M17 ;enable steppers
G91 ; Set relative for lift
G1 Z10 ; lift z by 10
G90 ;back to absolute positioning
G1 Z0.1;
M190 R27 ; set bed to 27 for cool-down
G4 S90 ; wait for temp stabilization
M190 R27 ;verify temp below threshold
G0 X200 Y235 ;move to back corner
G0 Y0 f9000;wipe forward
G0 Y235 ;wipe aft
G0 X144 Y235 ;move to mid bed aft
G0 Y0 f9000;wipe forward
G0 Y235 ;wipe aft
G0 X88 Y235 ;move to mid bed aft
G0 Y0 f9000;wipe forward
G0 Y235 ;wipe aft
G0 X32 Y235 ;move to mid bed aft
G0 Y0 F9000;wipe forward
G0 Y235 ;wipe aft
G1 Z10;
G28 ; home

smartin015, 2022-03-20:
Hey @sheldo26 - thanks for the report. At first glance your GCODE seems to be correct, so we'll need to dig into why your `G4` and `M190` commands aren't being observed.

Can you please make another (i.e. failing) attempt, then complete the "System Info" section from the [bug report template](https://github.com/smartin015/continuousprint/issues/new?assignees=&labels=bug&template=bug_report.md&title=) so I can look at the logs from when it fails to wait? I've copied the instruction below:

> **System info:**
> Follow the instructions [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) to generate an OctoPrint system bundle, and attach it to this issue.

Thanks

smartin015, 2022-03-21:
`octoprint.log` tells an interesting story - I found this pattern in multiple places:

```
2022-03-20 11:09:54,867 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: cp_bed_clearing_script.gcode, owner: None, user: None
2022-03-20 11:09:54,942 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-03-20 11:09:55,040 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2022-03-20 11:09:55,158 - octoprint.plugins.bedcooldown - INFO - Bed cooldown trigger is configured for when time left is <= 300s and completion is >= 90.00%
2022-03-20 11:09:55,159 - octoprint.plugins.bedcooldown - INFO - However, plugin is not currently enabled
2022-03-20 11:09:55,173 - octoprint.plugins.dashboard - INFO - Gcode not pre-processed by Dashboard. Processing now.
2022-03-20 11:09:55,179 - octoprint.plugins.dashboard - INFO - GcodePreProcessor started processing.
2022-03-20 11:09:55,197 - octoprint.plugins.dashboard - INFO - Gcode pre-processing done.
2022-03-20 11:09:55,198 - octoprint.plugins.dashboard - INFO - GcodePreProcessor found layers: 1
2022-03-20 11:09:55,198 - octoprint.plugins.dashboard - INFO - GcodePreProcessor found filament changes: 0
2022-03-20 11:09:55,198 - octoprint.plugins.dashboard - INFO - GcodePreProcessor saving layer count in file metadata
2022-03-20 11:09:55,447 - octoprint.filemanager.analysis - INFO - Analysis of entry local:cp_bed_clearing_script.gcode finished, needed 0.66s
2022-03-20 11:13:05,316 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-20 11:22:47,681 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 3940, in _readline
    ret = self._serial.readline()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 6605, in readline
    c = self.read(1)
  File "/home/pi/oprint/lib/python3.7/site-packages/serial/serialposix.py", line 596, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-20 11:22:47,693 - octoprint.util.comm - ERROR - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
2022-03-20 11:22:47,721 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline after error"
2022-03-20 11:22:47,738 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2022-03-20 11:24:16,723 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial connection"
2022-03-20 11:24:16,775 - octoprint.util.comm - INFO - Serial detection: Performing autodetection with 1 port/baudrate candidates: /dev/ttyACM0@115200
2022-03-20 11:24:16,776 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyACM0, baudrate 115200
2022-03-20 11:24:16,779 - octoprint.util.comm - INFO - Connecting to port /dev/ttyACM0, baudrate 115200
2022-03-20 11:24:16,804 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2022-03-20 11:24:16,808 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-03-20 11:24:16,833 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Operational"
2022-03-20 11:24:16,842 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-03-20 11:24:16,886 - octoprint.plugins.continuousprint - WARNING - on_printer_ready
2022-03-20 11:24:16,886 - octoprint.plugins.continuousprint - INFO - Printer ready; performing next action <function ContinuousPrintDriver._begin_next_available_print.<locals>.<lambda> at 0x689ffa98>
```

Do you know what's causing these serial errors? I'd suggest looking through the possible causes at https://faq.octoprint.org/serialerror if you aren't causing them (e.g. by unplugging/re-plugging in the USB cable or powering the printer off and on again). 

The log entries by `octoprint.plugins.bedcooldown` are also interesting - I'm not sure if the plugin does anything when it's "disabled", but you could also try uninstalling it to see if that helps. LMK if it does, because I'd like to understand how/why it's interfering here.

smartin015, 2022-03-30:
Hey @sheldo26, any luck fixing your printing issues? Can you tell me what worked/didn't work?

smartin015, 2022-04-03:
Since it's been about two weeks without any updates, I'm going to go ahead and assume you found a solution or moved on to something else. I'l close this issue out; please comment if you'd like it opened again.

closed 2022-04-03 by smartin015