smartin015/continuousprint #43: Bed Clearing Script Not Running

AriesPlaysNation, 2022-03-23:
**Describe the bug**
Bed Clearing script not running on Ender 3 Max. Code below:
M17 ;enable steppers
G91 ; Set relative for lift
G0 Z10 ; lift z by 10
G90 ;back to absolute positioning
M190 R25 ; set bed to 27 and wait for cooldown
M190 R25 ; set bed to 27 and wait for cooldown
M190 R25 ; set bed to 27 and wait for cooldown
M190 R25 ; set bed to 27 and wait for cooldown
M190 R25 ; set bed to 27 and wait for cooldown
M190 R25 ; set bed to 27 and wait for cooldown
M190 R25 ; set bed to 27 and wait for cooldown
M190 R25 ; set bed to 27 and wait for cooldown
G0 X150 Y300 ;move to mid bed aft
G0 Z1 ;come down to 1MM from bed
G0 Y0 ;wipe forward
G0 Y300 ;wipe aft
G29 ; home

**To Reproduce**
Steps to reproduce the behavior:
1. Hit + sign to add to queue.
2. Push Start Managing
3. Wait until first print is done
4. Bed should go down to 27 degrees (which is does by natural cooling. Printer doesn't show that it's actually set to 27)
5. Hours later, still nothing.

**Expected behavior**
At step 4 the bed should be set to 27 and wait for cooldown then run the bed clearing script.

**Screenshots**
[image]

**System info:**
[systeminfo.txt](https://github.com/smartin015/continuousprint/files/8334189/systeminfo.txt)

**Browser logs:**

**Additional context**

gdgtt, 2022-03-26:
Hello, I'm having the same issue on one of my printers after the update (the other printer is fine). I've tried uninstalling/reinstalling the plugin multiple times with no luck. I even reinstalled octoprint and started fresh but it still does not run the bed clearing script. I also tried installing a backup I just made from the working printer so all plugins and files would be the same, but still no luck. 

Is there somewhere I can download the previous version of continuous print? Thanks!

gdgtt, 2022-03-26:
Okay I found the old repository and was able to downgrade to 1.3.1. Everything is working again. I'll keep monitoring this thread and hopefully there is a solution in the future. 

https://github.com/Zinc-OS/continuousprint/releases

Thank you for all your work!

smartin015, 2022-03-28:
@AriesPlaysNation - could you please attach the full system bundle (zip file) as described [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887)? That'll give me a copy of the system logs, which I need to debug. Thanks :)

@gdgtt I can't really help you directly without a [system bundle](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) - there's way too much going on under the hood to know what's wrong from a couple sentences. I understand if you want to stick with your downgrade for now, but if you can I'd appreciate trying the plugin again and capturing the system bundle when it fails for you.  

I'm also really intrigued that one printer isn't working, but another is - can you please tell me the make/model of them, and any differences in setup you have between the two? Thanks!

AriesPlaysNation, 2022-03-28:
Yup, I'll get to this sometime by the weekend. I just started a 30 something hour print a couple hours ago.

smartin015, 2022-04-07:
@AriesPlaysNation - how are things coming along? :)

AriesPlaysNation, 2022-04-07:
Very sorry! Both of my printers been going 24/7 the last couple weeks and haven't had time to test it out.
I'll squeeze it in very soon with some small test prints that go quickly!

peterson-caleb, 2022-04-14:
This just started happening for me on two machines. Had been running awesome since you took over development of the plugin and even before that. The machine finishes
[octoprint-systeminfo-20220414135534.zip](https://github.com/smartin015/continuousprint/files/8489512/octoprint-systeminfo-20220414135534.zip)
[octoprint-systeminfo-20220414135505.zip](https://github.com/smartin015/continuousprint/files/8489513/octoprint-systeminfo-20220414135505.zip)
 the first queue print then just idles. Two bundles attached.

smartin015, 2022-04-16:
Thanks for the report!

I spent some time this weekend changing how the queue manager handles its state, and added a watchdog timer to kick the driver if it somehow misses/mis-handles an octoprint event. I'm waiting to push this to `rc` until I get the OK on #6 and can do a full release. I'm hopeful this (pending) change will address these intermittent issues - will post back here when it's ready to try out, with instructions.

smartin015, 2022-04-17:
Ok, the new "state machine" implementation is released as [1.6.0rc1](https://github.com/smartin015/continuousprint/releases/tag/1.6.0rc1) - you can help test it out by switching to the Release Candidate channel (details [here](https://community.octoprint.org/t/how-to-use-the-release-channels-to-help-test-release-candidates/402))

smartin015, 2022-05-04:
`1.6.0` is now mainline. Please reopen this if you continue to see issues in 1.6.0, but as I haven't heard any reports in a while I'm going to go ahead and consider this fixed.

AriesPlaysNation, 2022-05-19:
I'm finally caught up with orders since I got a couple new printers. I'll be testing this tomorrow/Saturday to see if works better now. It's been awhile so I need to reset everything up and attach my shovel/guard back onto the hotend

AriesPlaysNation, 2022-05-20:
Original issue is fixed and it does move on to the next part.
The bl touch drops an error but I don't think it's continuous print related because I get the same error randomly when quickly starting a next print when ones finished.