smartin015/continuousprint #49: Request: a way to send a job to octoprint from prusaslicer and have it automatically added to the queue

napter, 2022-04-11:
Currently is there any way I can send a job from prusaslicer and have it start a new queue - or if a queue is printing have it added to the queue? 

While continuous print is great it takes me out of my current process, which is to slice from prusaslicer and send to octoprint with the "upload and print" option - I don't open the octoprint interface at all. 

I want to simply send a number of items to print one after the other without opening octoprint. 

Ideally what am thinking is the following: A setting that says "always manage" so continuous print is always managing and every print is part of the queue. With that option selected, if a new gcode is sent from prusaslicer using "upload and print" and nothing is printing, it starts a new job with the gcode as the only item. If a job is in middle of printing , then it adds the new gcode to the queue to print next.

smartin015, 2022-04-11:
Looking briefly into prusaslicer, it seems they're [using the OctoPrint REST API](https://github.com/prusa3d/PrusaSlicer/blob/eddcd93e826a0e526e2a00fedaf136bdd25a1855/src/slic3r/Utils/OctoPrint.cpp#L131) for uploading and printing.

Hooking into this would be difficult - I can't see a way around the problem where adding multiple files with "start printing" enabled would probably fail with a 409 error as the printer would be active already (see [API doc](https://docs.octoprint.org/en/master/api/files.html#upload-file-or-create-folder)). It'd kind-of work to auto-enqueue uploaded files as long as you didn't send the start request until the end. Is that something you're OK with?

Another option would be adding compatibility into prusaslicer - they're [already set up for multiple host types](https://github.com/prusa3d/PrusaSlicer/blob/eddcd93e826a0e526e2a00fedaf136bdd25a1855/src/slic3r/Utils/PrintHost.cpp#L49), and I can expose an endpoint for adding jobs to the queue. This would allow for better control overall, but there are two things in the way there: (1) I don't use prusaslicer, so it's a bit out of my way to contribute, and (2) I don't know whether they'd accept a whole new host type (or even just a modification) to support a plugin of a host type they already support natively.

napter, 2022-04-11:
The 409 from octoprint is only if you attempt to upload a file with the same name as the one being printed. You can upload a file with a different name. There is an event for when a file is uploaded that the plugin can listen to.

I don't think it will be necessary to create your own endpoint. (But if you did want to do that I would recommend you make it compatible with the octoprint API so that there wouldn't be any change to prusaslicer / cura / etc. That way would be compatible with anything that currently supports the octoprint API)

smartin015, 2022-04-12:
Gotcha. It seems reasonable to me to auto-enqueue files by default in most cases, since it's easy enough to clear the queue. Renamed the FR since this can be done generically and not just for prusaslicer.

Are you interested in helping to implement this? I'm currently busy working on some other parts of this plugin (among other projects). It may be a couple weeks before I start on this.

smartin015, 2022-06-08:
With release of `2.0.0` imminent, just wanted to circle back here with some thoughts.

The new release changes the behavior of queue management - rather than automatically stopping management when the queue is depleted, it instead enters an "idle" state and waits for work. This basically acts as the "always manage" option you requested; you only need to click Start Managing once.

Because of the change to management, newly added prints are put in a job in "draft" mode. Jobs in this mode are not considered for printing until they are saved, so that the printer doesn't suddenly start printing a half-baked job. With #65 I intend to add an option to set the draft state of the job when adding a new file/set, but that doesn't help us for octoprint uploads where this isn't specified. For this reason I'm considering a selectable setting behavior for file upload:

* Do nothing - current behavior
* Create job in draft mode - equivalent to clicking `+` on a file
* Create printable job - equivalent to clicking `+` on a file, then saving the job. This is what you're asking for :)

smartin015, 2022-06-10:
Hey @napter - I added a new setting in [`v2.0.0rc4`](https://github.com/smartin015/continuousprint/releases/tag/2.0.0rc4) that auto-adds items to the queue when uploaded to OctoPrint. You can find it under "Behavior" in the CPQ settings section. Please give it a try :)

smartin015, 2022-06-18:
Okay, this feature is now in "production" (v2.0.0 main release). I've confirmed remote file upload can add a file straight to the queue and it starts printing. I think this is everything needed here, but please reopen / file a new issue if you see anything missing. Thanks again for the suggestion!

closed 2022-06-18 by smartin015