smartin015/continuousprint #53: Managed-Bed Cooldown Feature is not documented.

willindiana, 2022-04-21:
**Is your feature request related to a problem? Please describe.**
Having finished V1 of the managed bed cooldown feature. I see we have this shiny new documentation page.
I would like to add this feature to documentation for completeness' sake. 

**Describe the solution you'd like**
I would like to outine the following documented points.

- why this feature was developed
- how to use each input field
- possible use cases for this feature
- add pictures to aid in demonstration
- faq for possible issues

**Additional context**
I primarily wrote this feature so I can take care of this issue, feel free to assign it to me I'll be aiming to knock it out this weekend.

willindiana, 2022-04-21:
Self-reminder: Add descriptor for pre-bed cooldown script area with description on possible g-code cooldown commands or other external trigger events.

smartin015, 2022-04-21:
Thanks for raising this - please do add documentation for the managed bed cooldown feature! 

Ironically, the specifics of how to go about documenting is itself undocumented :) If you haven't already figured it out, check out [the tutorial on MkDocs](https://www.mkdocs.org/getting-started/) which is what I settled on for the docs site. Specifically, `mkdocs serve` will get you a local instance with hot-reloading on changes. When we've merged in the docs I'll need to run the deploy script before it shows up. 

Also as a side note: the `rc` branch contains some new code for user-contributed printer profiles and scripts (see [here](https://github.com/smartin015/continuousprint/tree/rc/continuousprint/data), also [docs](https://smartin015.github.io/continuousprint/gcode-scripting/#contributing)). If any of your cooldown scripting is generic enough to use by default (or specific enough to be needed by a certain model of printer) then you can contribute it there, and we can see about hooking up the cooldown gcode settings section so it can also be affected by these templates.

smartin015, 2022-04-21:
Also also - I refactored the driver code so it's now more of a state machine than an event handler. Your code should continue to work as currently implemented, but it may now be easier to move away from busy looping by adding a new state function between `_state_start_clearing` and `_state_clearing` if you were interested in doing some further tweaks/refinements. No problem if you wanted to stick to the current version, though.

willindiana, 2022-04-24:
Wrote out the docs for the feature https://github.com/smartin015/continuousprint/pull/55
here's my pull request.

willindiana, 2022-04-24:
I created an issue for me to work on the improvements you suggested. @smartin015 https://github.com/smartin015/continuousprint/issues/56

smartin015, 2022-04-27:
Thanks! I've deployed your PR to https://smartin015.github.io/continuousprint/managed-bed-cooldown/.

closed 2022-04-27 by smartin015