smartin015/continuousprint #56: Improved Bed Cooldown Event Management

willindiana, 2022-04-24:
**Is your feature request related to a problem? Please describe.**
I would like to improve the implementation of bed-cooldown so that I don't have a spinning while loop holding up a thread in octoprint while it is waiting for the bed to hit a temperature. 

**Describe the solution you'd like**
Use something like Util.RepeatedTimer to do the following.
1. Measure timeout progress.
2. Monitor bed temperature for threshold
3. Don't use an inefficient while loop that holds up threads.
4. Periodically log progress for debugging purposes. 

**Additional context**
When I initially wrote this feature I had tried using a thread respecting module like util.RepeatedTimer but I was facing issues with it not blocking the bed clear. I would like to take another swing at things with the latest changes and follow what was suggested originally. Here https://github.com/smartin015/continuousprint/issues/6

smartin015, 2022-04-27:
The new state-machine implementation in `driver.py` gets its `action()` method called every 5 seconds by `__init__.py` (using RepeatedTimer). If you add a new `_state_managed_cooldown()` in the driver, you can put your bed temperature checking code there. You can transition to the new state when it's time to use it by returning `self._state_managed_cooldown` from another state (note: no parenthesis, you're returning the method and not invoking it).

HTH

smartin015, 2022-07-03:
Saw some potentially cooldown-related issues in #79, so I went ahead and made the changes with some unit tests to help detect regression. They'll be out in the next release candidate (hoping to create one tomorrow, after some physical testing)

closed 2022-07-03 by smartin015