smartin015/continuousprint #58: Integration of CPQ with BedReady plugin

smartin015, 2022-04-29:
**Is your feature request related to a problem? Please describe.**

https://github.com/jneilliii/OctoPrint-BedReady is a new OctoPrint plugin that uses the `@BEDREADY` at-command to check the current webcam view of the bed and pauses the print if the bed appears to be occupied. It seems like a no-brainer to add this as an option when clearing the print bed using CPQ, when this plugin is installed.

**Describe the solution you'd like**

At-command logic is [here](https://github.com/jneilliii/OctoPrint-BedReady/blob/master/octoprint_bedready/__init__.py#L112) in BedReady. Recent changes to CPQ now allow for at-commands to be used in the settings scripts, so it should be pretty drop-in. 

I think it makes sense to add a wiki entry describing the setup process, to raise awareness that this is possible. 

**Describe alternatives you've considered**

Given plugin messages are already set up for when the bed is considered "blocked", it's possible to also understand that the print is paused because the bed isn't clear, and take some action (e.g. try to clear the bed again). I don't think this is necessary as a first attempt, though.

Also possible to automatically integrate with BedReady given the plugin messages and do this without needing user configuration, but this again seems like extra work that isn't strictly necessary at the moment.

**Additional context**

@jneilliii - just a heads-up I'm hoping to publish a quick integration guide between your BedReady plugin and ContinuousPrint. LMK if you're interested in contributing, have suggestions on further integration, or if you'd like me to ping you when it's ready.

jneilliii, 2022-04-29:
Yeah, seems to be pretty straight-forward as just documenting for now. Always open to any changes necessary for other plugins to integrate.

smartin015, 2022-07-13:
Documented, will publish on next release.

closed 2022-07-13 by smartin015