smartin015/continuousprint #62: How to debug?

TylonHH, 2022-05-08:
Hey,
I noticed some strange behavior.  For example, it prints just one file over and over instead of the queue. Suddenly.
Or another thing: I took round about untill the next file is loaded (and printed), even the bed clearing script. 

I wanna know how to debug. I enabled within OctroPrint under Logging the Log Level "octoprint.plugins.continuousprint", but I dont see any log like this.

What would you suggest?

smartin015, 2022-05-10:
It looks like in #64 you have down to debug logging for the continuous print plugin. When I start debugging, I use those log messages and the output of the console in the browser (F12 opens the console in Firefox and Chrome). 

If you know python and/or javascript, you can also follow the steps in the [contributing guide](https://smartin015.github.io/continuousprint/contributing/) to set up a development instance of continuous print that you can modify, but debugging in this way gets pretty involved.

FWIW I really do appreciate your bug reports :) it may just take a bit to get to them sometimes.

smartin015, 2022-06-08:
Closing this out with additional config / debugging documentation added to [contribution guide](https://smartin015.github.io/continuousprint/contributing/)

closed 2022-06-08 by smartin015