smartin015/continuousprint #64: two jobs not queued

TylonHH, 2022-05-08:
Hi,

I added two files to the queue. But it's only one file printing multiple times.

Here is the log:
```
2022-05-08 17:23:06,297 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[]
2022-05-08 17:23:21,720 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[]
2022-05-08 17:23:36,854 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[]
2022-05-08 17:23:43,902 - octoprint.util.comm - INFO - Finished in 2453.857 s.
2022-05-08 17:23:43,903 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2022-05-08 17:23:43,946 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: E_V3_newest_BOTTOM_vase_mode.aw.gcode, owner: None
2022-05-08 17:23:45,116 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2022-05-08 17:23:52,501 - octoprint.plugins.continuousprint - DEBUG - TICK, IDLE, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[]
2022-05-08 17:23:52,501 - octoprint.plugins.continuousprint - INFO - _state_printing -> _state_success
2022-05-08 17:23:57,517 - octoprint.plugins.continuousprint - DEBUG - TICK, IDLE, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[]
2022-05-08 17:24:08,093 - octoprint.plugins.continuousprint - INFO - _state_success -> _state_start_clearing
2022-05-08 17:24:13,106 - octoprint.plugins.continuousprint - DEBUG - TICK, IDLE, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[]
2022-05-08 17:24:13,224 - octoprint.plugins.continuousprint - INFO - Wrote file cp_bed_clearing_script.gcode
2022-05-08 17:24:13,234 - octoprint.filemanager.analysis - INFO - Starting analysis of local:cp_bed_clearing_script.gcode
2022-05-08 17:24:13,274 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/pi/.octoprint/uploads/cp_bed_clearing_script.gcode
2022-05-08 17:24:13,283 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: cp_bed_clearing_script.gcode, owner: None, user: None
2022-05-08 17:24:13,411 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2022-05-08 17:24:13,453 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: cp_bed_clearing_script.gcode, owner: None, user: None
2022-05-08 17:24:14,196 - octoprint.plugins.continuousprint - INFO - _state_start_clearing -> _state_clearing
2022-05-08 17:24:14,199 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-05-08 17:24:14,226 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2022-05-08 17:24:17,353 - octoprint.util.comm - INFO - Finished in 3.970 s.
2022-05-08 17:24:17,354 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2022-05-08 17:24:17,403 - octoprint.filemanager.analysis - INFO - Starting analysis of local:cp_bed_clearing_script.gcode
2022-05-08 17:24:17,496 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: cp_bed_clearing_script.gcode, owner: None
2022-05-08 17:24:17,653 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/pi/.octoprint/uploads/cp_bed_clearing_script.gcode
2022-05-08 17:24:19,208 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=cp_bed_clearing_script.gcode, materials=[]
2022-05-08 17:24:19,501 - octoprint.filemanager.analysis - INFO - Result is empty, no extrusions found
2022-05-08 17:24:19,502 - octoprint.filemanager.analysis - INFO - Analysis of entry local:cp_bed_clearing_script.gcode finished, needed 2.10s
2022-05-08 17:24:24,211 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=cp_bed_clearing_script.gcode, materials=[]
2022-05-08 17:24:29,215 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=cp_bed_clearing_script.gcode, materials=[]
2022-05-08 17:24:34,219 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=cp_bed_clearing_script.gcode, materials=[]
2022-05-08 17:24:35,736 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2022-05-08 17:24:39,225 - octoprint.plugins.continuousprint - DEBUG - TICK, IDLE, path=cp_bed_clearing_script.gcode, materials=[]
2022-05-08 17:24:39,226 - octoprint.plugins.continuousprint - INFO - _state_clearing -> _state_start_print
2022-05-08 17:24:44,234 - octoprint.plugins.continuousprint - DEBUG - TICK, IDLE, path=cp_bed_clearing_script.gcode, materials=[]
2022-05-08 17:24:50,039 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: E_V3_newest_BOTTOM_vase_mode.aw.gcode, owner: None, user: None
2022-05-08 17:24:50,093 - octoprint.plugins.continuousprint - INFO - E_V3_newest_BOTTOM_vase_mode.aw.gcode
2022-05-08 17:24:50,101 - octoprint.util.comm - INFO - Starting job on behalf of plugin continuousprint
2022-05-08 17:24:50,137 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2022-05-08 17:24:50,163 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: E_V3_newest_BOTTOM_vase_mode.aw.gcode, owner: None, user: None
2022-05-08 17:24:50,807 - octoprint.plugins.continuousprint - INFO - _state_start_print -> _state_printing
2022-05-08 17:24:50,821 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-05-08 17:24:50,893 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2022-05-08 17:24:55,831 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[]
2022-05-08 17:25:09,851 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[]
2022-05-08 17:25:26,063 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[]
```

And here is the queue:
[image]

I also wonder why its still clearing...
The next (same) job is already printing.

kind regards

TylonHH, 2022-05-08:
Why?
[image]

smartin015, 2022-05-10:
Did you re-order the prints at all after you added them? I suspect there may be a bug in the json storage implementation that may not properly reorder the jobs. For better or worse, this will be totally replaced in #60.

sidneyns, 2022-05-13:
I am having the same issue, when I place new files or I add more quantity when the printer is printing and with a queue , it wont print after finish the original queue

sidneyns, 2022-06-02:
> I've released 2.0.0rc1, which is basically a full overhaul of the queuing
> logic at issue here. @sidneyns <https://github.com/sidneyns> / @TylonHH
> <https://github.com/TylonHH>, if you're able to consistently reproduce
> this problem, can you try it in the new version? Thanks!

will try my friend

TylonHH, 2022-06-04:
I test this newest RC on two maschine, so far it seems running.
Things I noticed so far:
a) I noticed, that suddenly the clearing script was not running, Instead the next print was starting. I could handle it to remove the print by hand. How can I log this?
b) Now I try one job where one print is set Zero (the amount), just to keep the file in the list, but not to print. Then the plugin is stuck in zoomed-in animation
c) is there a zooming animation or something? When I save a job or update the numbers of print the text of the plugin 100% on the screen and zooms out to normal, its very quick, but annoying.

The new layout is a big step. Good work

TylonHH, 2022-06-04:
d) the function to save jobs is awesome. When I edit a job and save (again) with the same name I would expect an question to override or to save under a new name. Now is saved under a new name with ....(1)

TylonHH, 2022-06-04:
Big importand question:
When I save a job, a new folder is created. Inside a the two files. Are they a copy of this file? This would be bad.
Because when I use a file in different jobs and I wanna make a change inside the (main) file, I expect that this change is in every job where this file is used.

TylonHH, 2022-06-04:
Now I have one job with two files, where just one file is printed. The second seems to be ignored.
[image]

Here is the log:
```
2022-06-04 19:23:51,618 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=E_V3_LID_vase_mode_deeper_notch_Frage_fixed100-5.aw.gcode, materials=[]
2022-06-04 19:23:56,629 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=E_V3_LID_vase_mode_deeper_notch_Frage_fixed100-5.aw.gcode, materials=[]
2022-06-04 19:24:00,824 - octoprint.util.comm - INFO - Finished in 2228.685 s.
2022-06-04 19:24:00,825 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2022-06-04 19:24:00,842 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: E_V3_LID_vase_mode_deeper_notch_Frage_fixed100-5.aw.gcode, owner: None
2022-06-04 19:24:01,272 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2022-06-04 19:24:01,638 - octoprint.plugins.continuousprint - DEBUG - TICK, IDLE, path=E_V3_LID_vase_mode_deeper_notch_Frage_fixed100-5.aw.gcode, materials=[]
2022-06-04 19:24:01,835 - octoprint.plugins.continuousprint - INFO - Wrote file cp_bed_clearing_script.gcode
2022-06-04 19:24:01,844 - octoprint.filemanager.analysis - INFO - Starting analysis of local:cp_bed_clearing_script.gcode
2022-06-04 19:24:01,865 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: cp_bed_clearing_script.gcode, owner: None, user: None
2022-06-04 19:24:01,869 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/pi/.octoprint/uploads/cp_bed_clearing_script.gcode
2022-06-04 19:24:02,074 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2022-06-04 19:24:02,088 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: cp_bed_clearing_script.gcode, owner: None, user: None
2022-06-04 19:24:02,799 - octoprint.plugins.continuousprint - INFO - _state_start_clearing -> _state_clearing
2022-06-04 19:24:02,824 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-06-04 19:24:02,861 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2022-06-04 19:24:02,969 - octoprint.plugins.continuousprint - DEBUG - Refreshing UI state
2022-06-04 19:24:04,882 - octoprint.util.comm - INFO - Finished in 2.897 s.
2022-06-04 19:24:04,883 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2022-06-04 19:24:04,889 - octoprint.filemanager.analysis - INFO - Starting analysis of local:cp_bed_clearing_script.gcode
2022-06-04 19:24:04,998 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: cp_bed_clearing_script.gcode, owner: None
2022-06-04 19:24:05,057 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/pi/.octoprint/uploads/cp_bed_clearing_script.gcode
2022-06-04 19:24:05,060 - werkzeug - INFO - 127.0.0.1 - - [04/Jun/2022 19:24:05] "GET /?action=snapshot HTTP/1.1" 200 -
2022-06-04 19:24:06,777 - octoprint.filemanager.analysis - INFO - Result is empty, no extrusions found
2022-06-04 19:24:06,778 - octoprint.filemanager.analysis - INFO - Analysis of entry local:cp_bed_clearing_script.gcode finished, needed 1.89s
2022-06-04 19:24:07,982 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=cp_bed_clearing_script.gcode, materials=[]
2022-06-04 19:24:07,983 - octoprint.plugins.continuousprint - INFO - Clearing bed
2022-06-04 19:24:08,121 - octoprint.plugins.continuousprint - DEBUG - Refreshing UI state
2022-06-04 19:24:13,135 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=cp_bed_clearing_script.gcode, materials=[]
2022-06-04 19:24:18,145 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=cp_bed_clearing_script.gcode, materials=[]
2022-06-04 19:24:22,741 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2022-06-04 19:24:23,151 - octoprint.plugins.continuousprint - DEBUG - TICK, IDLE, path=cp_bed_clearing_script.gcode, materials=[]
2022-06-04 19:24:23,819 - octoprint.plugins.continuousprint - DEBUG - Refreshing UI history
2022-06-04 19:24:23,842 - octoprint.plugins.continuousprint - INFO - Attempting to print E_V3_LID_vase_mode_deeper_notch_Frage_fixed100-5.aw.gcode (sd=False)
2022-06-04 19:24:23,868 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: E_V3_LID_vase_mode_deeper_notch_Frage_fixed100-5.aw.gcode, owner: None, user: None
2022-06-04 19:24:23,946 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2022-06-04 19:24:23,955 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: E_V3_LID_vase_mode_deeper_notch_Frage_fixed100-5.aw.gcode, owner: None, user: None
2022-06-04 19:24:24,500 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-06-04 19:24:24,525 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2022-06-04 19:24:24,699 - octoprint.plugins.continuousprint - DEBUG - Refreshing UI state
[... cut ...]
2022-06-04 19:24:29,944 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=E_V3_LID_vase_mode_deeper_notch_Frage_fixed100-5.aw.gcode, materials=[]
2022-06-04 19:24:29,945 - octoprint.plugins.continuousprint - INFO - Printing
2022-06-04 19:24:30,034 - octoprint.plugins.continuousprint - DEBUG - Refreshing UI state
2022-06-04 19:24:35,062 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=E_V3_LID_vase_mode_deeper_notch_Frage_fixed100-5.aw.gcode, materials=[]
2022-06-04 19:24:40,073 - octoprint.plugins.continuousprint - DEBUG - TICK, BUSY, path=E_V3_LID_vase_mode_deeper_notch_Frage_fixed100-5.aw.gcode, materials=[]
```

smartin015, 2022-06-04:
Thanks for all the testing!

> a) I noticed, that suddenly the clearing script was not running, Instead the next print was starting. I could handle it to remove the print by hand. How can I log this?

I'm not sure why this is happening. Could you check the plugin settings and confirm your clearing script is still there? This happens every time a print finishes?

> b) Now I try one job where one print is set Zero (the amount), just to keep the file in the list, but not to print. Then the plugin is stuck in zoomed-in animation

Good find - I did not think about the use case for 0-count Sets, so doing this errors. I'll make a note to fix this behavior, and also the error behavior (it doesn't exit a "loading" state).

> c) is there a zooming animation or something? When I save a job or update the numbers of print the text of the plugin 100% on the screen and zooms out to normal, its very quick, but annoying.

There is no zooming animation - I suspect another of your plugins is interacting with my UI changes. Can you try CPQ with all other plugins disabled to confirm? If it is a plugin conflict and you can tell me which one is the cause, I can edit the page to get rid of the conflicting CSS.

> d) the function to save jobs is awesome. When I edit a job and save (again) with the same name I would expect an question to override or to save under a new name. Now is saved under a new name with ....(1)

This behavior matches the download behavior of browsers, which is to add a suffix to prevent naming conflicts. You can rename the file with the scissors icon after it's downloaded. If you want to replace a `.gjob`, you can delete it before you save the new one.

> When I save a job, a new folder is created. Inside a the two files. Are they a copy of this file? This would be bad.
Because when I use a file in different jobs and I wanna make a change inside the (main) file, I expect that this change is in every job where this file is used.

`.gjob` files are archives - they store a copy of all GCODE files. When you add a `.gjob` back to the queue, it extracts the GCODE into a folder with the same name as the file, so that OctoPrint can run those files. This is important for portability - you can save the .gjob to your computer, delete everything on your OctoPrint, and still upload and print the `.gjob` with no missing files. It's also important for LAN queues (a printer has all the files it needs to run a print job by fetching the `.gjob` file over the network).

I'm not sure I understand your use case here. Are you editing GCODE files by hand?  

> Now I have one job with two files, where just one file is printed. The second seems to be ignored.

I think what you want to do here is set your `E_V3...` files to a count of `1`, and your `Job 1` to a count of `50`. This will print each item, then loop back and print each item again. You'll still have 50 of each printed at the end.

TylonHH, 2022-06-04:
> I'm not sure why this is happening. Could you check the plugin settings and confirm your clearing script is still there? This happens every time a print finishes?

As it happens just ones I'll like to ignore it for now

> There is no zooming animation - I suspect another of your plugins is interacting with my UI changes. Can you try CPQ with all other plugins disabled to confirm? If it is a plugin conflict and you [...]

When it's not CP related don't care about it

> `.gjob` files are archives - they store a copy of all GCODE files. When you add a `.gjob` back to the queue, it extracts the GCODE into a folder with the same name as the file, so that OctoPrint can [...]

Good to know. So it's not for my workflow.

> I'm not sure I understand your use case here. Are you editing GCODE files by hand?

Yes. I just make smal improvements like: temperature, global offsets, homing after prints (so I don't have to care about the 5s delay between the print and the clearing ;-) )

> I think what you want to do here is set your `E_V3...` files to a count of `1`, and your `Job 1` to a count of `50`. This will print each item, then loop back and print each item again. You'll still [...]

Got ya.

smartin015, 2022-06-08:
I created another prerelease `2.0.0rc2` to include the 0-count and better error handling changes - @TylonHH, please give it a try!

This time, you'll need to clear out the plugin data (`Settings > Plugin Manager > the little eraser icon next to the CPQ plugin`) in order for the DB to be updated and allow you to set `count=0` for jobs/sets.

TylonHH, 2022-06-13:
So far the RC4 is running well

smartin015, 2022-06-14:
Great! I expect I'll release it on the 16th if nobody finds any bugs caused by v2.0.0 changes.

smartin015, 2022-06-18:
Ok, release complete. Please open a new bug if you find other strange behavior - thanks!

closed 2022-06-18 by smartin015