smartin015/continuousprint #65: Add to queue through API

aburtonProto, 2022-05-10:
**Is your feature request related to a problem? Please describe.**
"DEPRECATED: plugin/continuousprint/add is no longer used and will be removed in the next major release."

**Describe the solution you'd like**
An alternative way to add a file to the queue through API

**Additional context**
We are remotely pushing files to clients and want to be able to add them to the queue as well.  This worked previously, but as noted the method is now deprecated.  
Below is the code we have used to send the files previously.
```
def send_print_job(clinic, printer, gcode, attachment_name):
    secret = _get_secret(f"somethingVagueThatWasRemovedForGitHub")
    base_url = printer["Url"]
    if not base_url.startswith("https"):
        base_url = f"https://{base_url}"
    if not base_url.endswith("/"):
        base_url += "/"

    print(f"- Making request to {base_url + ADD_FILE_ENDPOINT}")
    resp = requests.post(
        base_url + ADD_FILE_ENDPOINT,
        auth=(printer['Name'], secret['ngrok']),
        headers={'X-Api-Key': secret['api']},
        files = [ ('file', (attachment_name, gcode, 'application/octet-stream')) ]
    )
    print(resp)
    print(resp.json())
    if resp.status_code >= 400:
        raise Exception(f"Error, could not add file: {resp.content}")

    print(f"- Making request to {base_url + ADD_QUEUE_ENDPOINT}")
    resp = requests.post(
        base_url + ADD_QUEUE_ENDPOINT,
        auth=(printer['Name'], secret['ngrok']),
        headers={'X-Api-Key': secret['api']},
        data={"items": json.dumps([{"name":attachment_name,"path":attachment_name,"sd":False}])}
    )
    print(resp)
    print(resp.json())

    if resp.status_code >= 400:
        if resp.status_code == 404:
            raise Exception("Error, running old version of OctoPrint. Need to update")
        raise Exception(f"Error, could not add to queue: {resp.content}")
```

The endpoints are 
```
ADD_FILE_ENDPOINT  = "api/files/local"
ADD_QUEUE_ENDPOINT = "plugin/continuousprint/add"
```

smartin015, 2022-05-10:
Thanks for this feedback - that specific implementation of file adding is indeed going away in `2.0.0`, but we can still support this use case, possibly even with less client side code needed. Would you prefer:

1. A similar "add file to queue" endpoint as what you're using now, optionally with additional params to allow for adding whole jobs (with multiple print files / counts, material specification etc.)?
2. A change in the default behavior that auto-creates a new job in the queue whenever a print is uploaded via the octoprint API? (see #49)

In any case, it'll still be a few days before I have `2.0.0` in a good state to deploy to the beta channel. We'll be able to confirm everything's working well before it goes live.

aburtonProto, 2022-05-12:
I vote for number 1.
If we are adding extra parameters, a useful one would be priority level.  For example to add to the front of the queue or the back.  But just being able to add to the queue programmatically is the important part for me.

smartin015, 2022-06-02:
Okay - after quite a few changes, I've pulled the backend handlers out of the `__init__.py` and into an `api.py` - should be a bit clearer for future integration.

The current "add to queue" endpoint is [/set/add](https://github.com/smartin015/continuousprint/blob/rc/continuousprint/api.py#L145). This takes `name`, `path`, `sd`, `count`, `jobName` and `job` params - allowing for a new job to be created (with `jobName` as its name) if `job` is empty.

An example of usage:

```
data = {
    'path': 'sample-cube-belt.gcode',
    'sd': 'false',
    'count': '1',
    'jobName': 'Job 1',
    'job': '',
}
response = requests.post('http://hostname/plugin/continuousprint/set/add', ... data=data)
```

@aburtonProto, is that what you're looking for? If it's sufficient, I can clean it up a bit, e.g. remove the need to specify an empty `job`, provide default `sd=False`.  "Priority level" is a bit iffy at the moment as it uses arbitrary float values for queue items, so I'm inclined to leave it at that for now and improve later.

aburtonProto, 2022-06-03:
Looks great.  I'll test this out and report back.  Thank you!

smartin015, 2022-06-10:
The [`v2.0.0rc4`](https://github.com/smartin015/continuousprint/releases/tag/2.0.0rc4) candidate release improves this a bit:

* `'job'` no longer needed
* `'jobDraft'` field now exists, default `True`. - set to `False` to enable immediate printing upon submission
* Integer fields can now be actual integers instead of strings
* Updated the [API doc](https://smartin015.github.io/continuousprint/api/)
* Updated the [`example.py`](https://github.com/smartin015/continuousprint/blob/2.0.0rc4/api_examples/example.py) reference implementation

@aburtonProto PTAL - I'll close this issue for now since I think this is the last work that was needed. Please comment/reopen if there's anything I've missed. Thanks!

closed 2022-06-10 by smartin015