smartin015/continuousprint #70: Request: Add a way to remove slicer "end gcode" when using continuous print

napter, 2022-06-01:
My standard end GCode has some codes like turn off steppers, turn off bed. However, when doing continuous print I don't want to run those, I can manage the end gcode by adding it to the script sections in continuous print.

It would be nice if there was a way to remove that gcode when using continuous print. A couple of different ways this can be implemented:
* Remove everything after a certain comment/tag in the gcode. Then I will add that at the top of my end gcode
* Remove any line containing some unique text (this can be placed as a comment on the end of the line)
* Remove everything that matches one or more regex matching expressions.

smartin015, 2022-06-02:
Hey napter - neat idea! I've had gcode inspection in the back of my mind for a while now for a couple different reasons. I'm particularly interested in it for #35, where it would be a big improvement to inspect the file and confirm it's sliced for the printer it's about to run on.

One thing I've not yet managed to design around is the substantial differences between the outputs of different slicers. I'd like to offer something that doesn't require the user to customize their slicer, lock them into a particular slicer implementation, or require them to develop a complex regex. 

I'm considering creating a different gcode analyzer plugin that can identify different parts of the gcode file and extract information about the printer it's intended to run on. That plugin could have user-contributed parsers for different slicers which would provide a common output that CPQ could then use.

As a result, we could pre-process gcode files and remove any unnecessary start/end steps, but it would require this additional plugin to be installed alongside CPQ.

WDYT?

napter, 2022-06-02:
I agree about avoiding regex. However, a simple macro using a comment or "@" like octoprint does would not be slicer specific. So whichever slicer you are using you add something like this to tell continuous print to remove the block:

@CP-REMOVE-START
//your start or ending gcode that should be skipped 
@CP-REMOVE-END

That should be a lot simpler than trying to create and maintain a gcode parser.

Are you on discord? I'd love to chat: I'm CookiecadNathan#8205

smartin015, 2022-06-03:
Sure - sent you a friend request on Discord.

WRT adding tags to the gcode - that means you'd need to change/customize your slicer settings, which is brittle to switching between slicers (or computers, if not using a cloud slicer).

I use Kiri:Moto, which already tags gcode files with `; --- shutdown ---` for the ending script section and has a header that clearly indicates the origin of the script. If we're customizing something already, might as well customize a plugin to read these existing tags so that every user of that slicer benefits.

Could potentially also have a dumb failover to detect `M104` (hotend off), `M140` (bed off) and other similar commands that hint at the end of the gcode file, or just start at the end of the file and backtrack until consecutive `G1/G0` commands are detected (i.e. finishing the last move). Either/both of these can be written once and used indefinitely.

napter, 2022-06-03:
"WRT adding tags to the gcode - that means you'd need to change/customize your slicer settings, which is brittle to switching between slicers (or computers, if not using a cloud slicer)."

I don't think that's an issue. If someone has end-gcode then they have to deal with moving it from computer to computer anyway. This isn't specific to to this process.

"Could potentially also have a dumb failover to detect M104 (hotend off), M140 (bed off) and other similar commands that hint at the end of the gcode file"

This could work - just a list of commands to remove, I'm mostly looking to remove turning off the hot end, bed and motors.

smartin015, 2022-06-04:
Let's consider the strip-commands option first then. WDYT about just a check-box for exactly what you're asking, i.e. prevent turning off hot end(s), bed, and motors in the print file? 

Will have to look into whether it's possible to intercept OctoPrint GCODE commands being run, or if the plugin would have to pre-process and export a secondary .gcode file with the lines removed.

napter, 2022-06-05:
I think that would do the trick

DrewCPU, 2022-12-14:
I'm in a similar boat. I have a GCODE script set to run at the start of every print that runs my bed-leveling. Normally, I prefer putting it in OctoPrint instead of my slicer because it's easier to change without affecting old models I want to reprint. Problem is that OctoPrint appends that script to the beginning of my cleaning script, which causes is to level the bed before it's cleared the print. I'll get in the habit of removing the scripts before I start continuously printing, but it would be great if there were a way to bypass the defined scripts in OctoPrint.

smartin015, 2022-12-20:
@DrewCPU - thanks for the feedback. This is actually a different (but still important) problem regarding the `GCODE Scripts` section of OctoPrint running on every event, not every print file. I'm beginning to suspect that I need to patch the core octoprint behavior so that this happens more as expected, i.e. not on clearing operations. 

Note that the latest RC version of ContinuousPrint now allows for adding a script to fire at the start of a queued print. This doesn't help with running out of band of CPQ, but it would work for queued prints to move your startup scripts there instead.

smartin015, 2022-12-29:
Okay, after looking into this a bit I think I have a way to do this by hooking into OctoPrint. 

Inside `util/comm.py` there's a `MachineCom._get_next_from_job` method which grabs lines from the currently active gcode file. I can pull a trick similar to #180 that patches this method and calls the original repeatedly until a "good" line or the end of the job is reached.

There's *probably* not too much risk in patching this, assuming proper safeguards for exception handling etc. are followed, since it was last changed meaningfully ~4 years ago. 

Failure mode would be these commands running again, worst case of which could be cold extrusion/move failures on a print - but that should get caught by printer firmware checks at this point.

@napter - I'll stage a PR to go in after `v2.3.0` comes out in a few days and ping this issue when I have an RC  for you to try.

smartin015, 2023-03-08:
This is now live in https://github.com/smartin015/continuousprint/releases/tag/2.4.0

PR #181 (2022-12-31): Configurable GCODE override in print files
Fixes #70 by adding configuration that allows ignoring certain GCODE lines when printing through a .gcode file.
closed 2023-03-08 by smartin015