smartin015/continuousprint #71: Request: An option to run the bed clearing script at the end of the print

napter, 2022-06-03:
I keep making tweaks to my bed clearing script and each time I have to copy/paste it to the final gcode since it doesn't clear the bed at the end of the print. I am not sure if there is any case where someone wants to clear the bed in between prints but doesn't want to at the end - personally I want the bed cleared after every print including the last one. I suggest either making it an option, or always running the clearing script at the end.

smartin015, 2022-06-04:
Good idea.

With a belt printer I also find myself missing the ability to clear the bed at the start of the first print as well, just in case there's something already on the bed. I imagine this as being two checkboxes in the settings page.

This'd be a couple weeks out at the earliest - I need to let `2.0.0rc1` soak for a bit to shake out any issues before a full deploy.

smartin015, 2022-11-16:
Just submitted #149, which enables this. When it goes live (i.e. when `v2.3.0` is released) it'll be possible to assign multiple scripts to run on any CPQ event.

closed 2022-11-16 by smartin015