smartin015/continuousprint #74: Failure Recovery with The Spaghetti Detective aka obico (own server)

TylonHH, 2022-06-13:
**Describe the bug**
obico is not recognized

**To Reproduce**
install the newest obico plugin

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Since TSD is sold to obico, the plugin was updated. I'm not sure if it's importand to know, I run obico on an own server (since 4 printers 24/7 is to expensive).

Mayve the link just has to update?!

cheers

smartin015, 2022-06-14:
Thanks for the report - this is probably because of some OctoPrint event objects that were prefixed with "spaghetti detective" and probably now prefixed with "obico". Since this is not a bug relating to new features in v2.0.0, I'm going to wait to fix this until after the major release, just in case a fix would cause bugs elsewhere.

smartin015, 2022-07-03:
The Obico fix is now available in [v2.1.0rc1](https://github.com/smartin015/continuousprint/releases/tag/2.1.0rc1) (note: release candidate, not yet in the "official" channel). @TylonHH please give the new version a try and let me know if you have any more issues.

I'm going to close this as fixed, but we can reopen if there's still a problem. Thanks again for letting me know there was a problem here, so I can fix it :)

closed 2022-07-03 by smartin015