smartin015/continuousprint #78: keep getting Continuous Print API (Error 500) INTERNAL SERVER ERROR

sidneyns, 2022-06-19:
hello 

I have a couple of octopi installed on several raspberry pis and one of them keeps showing this error.  I have alredy tried to unistall and install several times, but no success 

any clues?

smartin015, 2022-06-20:
Hey Sidney - thanks for the report, and sorry the upgrade wasn't a smooth one.

If you haven't already, you might try making a backup of the GCODE scripts in the plugin, then deleting plugin data using the eraser button in the plugin manager settings page:

[image]

But if you're able to before that - click the "System Info" button at the bottom of the settings dialog, then "Download Systeminfo Bundle" and attach it here. That contains various logs we can use to understand where the error is coming from.

smartin015, 2022-06-28:
Hey @sidneyns, are you still having problems with this?

sidneyns, 2022-06-28:
Hello my friend.. the new version fixed the issue

thank you

Sidney

smartin015, 2022-06-28:
Great, thanks!

closed 2022-06-28 by smartin015

craftking7, 2023-09-06:
Hi, i get the same error code when i try to start a print. I've already tried the method described here, but it didn't help. Any ideas?
[octoprint-systeminfo-20230906104759.zip](https://github.com/smartin015/continuousprint/files/12536539/octoprint-systeminfo-20230906104759.zip)