smartin015/continuousprint #79: CPQ is managing but nothing printing

napter, 2022-06-20:
I left this printing and it got stuck in this state. 
[image]

CPQ seems like it should move to the next but it is just idle. Is there a log I can look at to see what happened?

napter, 2022-06-20:
It was v2.0.0. I had to downgrade to 1.6 to get it to work. 

I think both 1.6 and 2.0.0 have a bug where they get out of sync with the octoprint status - so the plugin is waiting for octoprint to be available - but octoprint is idle. I don't know for sure if that's the cause but it is my theory. When this happens the CPQ says managing but nothing happens and the only way that I've found to fix it is to restart octoprint.

smartin015, 2022-06-20:
I see. I've experienced stuckness before and fixed it with a restart, but I haven't had it happen since the timer fix in rc4... sadly, I don't think there's a way to debug this further without some logs to inspect. 

It *should* be interpreting "operational" status as the printer being idle, and moving on from the printing state (see [here](https://github.com/smartin015/continuousprint/blob/master/continuousprint/driver.py#L196)). The only thing I can think of that would prevent this is if the state stops getting updated, i.e. the watchdog timer breaks and stops repeating, but all exceptions are caught [here](https://github.com/smartin015/continuousprint/blob/master/continuousprint/__init__.py#L225) (this was the rc4 fix).

Let's both keep an eye out for another failure like this, and capture logs when it happens. With any luck, we can find a root cause then.

winchester1094, 2022-06-28:
I am having the same issue. Prints the first file from SD and then sits idle. I have tried using m190 and bed cool down (I prefer the cool down) and when the first print is complete the cpq says waiting for printer to be ready. It never starts to run any of the cool down script.

winchester1094, 2022-06-29:
> Hi @winchester1094 - please take a look at the [comment](https://github.com/smartin015/continuousprint/issues/79#issuecomment-1160408232) upthread. We need a sysinfo bundle from when the printer is stuck in order to debug this.
[octoprint-systeminfo-20220629021721.zip](https://github.com/smartin015/continuousprint/files/9006725/octoprint-systeminfo-20220629021721.zip)

smartin015, 2022-07-01:
Interesting - @napter, I looked at your logs and I do see that it stayed in cooldown mode for around 6 hours, when the default should be just 60 minutes. I also need to reduce the logspam from being in the cooling state.

@winchester1094, yours also looks to be bed cooldown related, but there's actually an exception thrown:

```
bed_cooldown_script = self._get_key(Keys.BED_COOLDOWN_SCRIPT).split("\n")
NameError: name 'Keys' is not defined
```

The bed cooldown seems to be a bit of a blind spot for me, as I don't use it on my own setup. I'll make it a priority to fix these bugs and write up some unit tests to prevent any more regressions here. Will post back here when there's a release candidate ready, in the meantime see if you can make do without the bed cooldown option for now. Thanks again for the logs!

winchester1094, 2022-07-01:
I also tried not using the bed cool down. It never made it to the bed clearing script either way. I have reverted back to 1.6 but haven't attempted to use it again yet. Thanks

smartin015, 2022-07-03:
Okay, I just published [v2.1.0rc1](https://github.com/smartin015/continuousprint/releases/tag/2.1.0rc1) in the RC channel - it fixes the obvious errors I found in the logs, but to your point @winchester1094 there may be some other issue we haven't discovered yet. 

Can you and @napter try the new release candidate and let me know if the issues persist? If there are still issues, I'd appreciate another sysinfo dump when running the new version so I can debug further. Thanks!

smartin015, 2022-07-11:
Hey @winchester1094, @napter - have either of you had time to try your luck with the new release candidate?

winchester1094, 2022-07-11:
@smartin015 I have not. I'll install and run it soon. Thanks

napter, 2022-07-15:
@smartin015 I just tried it. It seems it tried moving on to the next print without clearing the bed first. The log shows "state clearing" but I'm pretty sure it didn't run the clear bed gcode.

2022-07-15 06:09:07,728 - octoprint.plugins.continuousprint - INFO - Timeout of 15 minutes exceeded
2022-07-15 06:09:07,728 - octoprint.plugins.continuousprint - INFO - _state_cooldown -> _state_clearing
2022-07-15 06:09:12,774 - octoprint.plugins.continuousprint - INFO - Attempting to print Cookiecad Cutter - 2022-07-14T001815.912-A1 (Prusa).gcode (sd=False)

smartin015, 2022-07-16:
> @smartin015 I just tried it. It seems it tried moving on to the next print without clearing the bed first. The log shows "state clearing" but I'm pretty sure it didn't run the clear bed gcode.

Right you are - I somehow managed to implement all the state transitions without actually calling the clear bed script... oops. Thanks for catching this.

I fixed the bug, added tests asserting the script is called, and ran through a virtual print with an `@pause` command to verify. You should be able to upgrade to [`2.1.0rc2`](https://github.com/smartin015/continuousprint/releases/tag/2.1.0rc2) now on the RC channel and see the right behavior.

winchester1094, 2022-07-17:
@smartin015 With 2.1.0rc2 I am unable to use files on my sd card. It loads it to the queue but when I click start managing it says file missing will be skipped

smartin015, 2022-07-18:
> @smartin015 With 2.1.0rc2 I am unable to use files on my sd card. It loads it to the queue but when I click start managing it says file missing will be skipped

I see where the problem is and will work on a fix and some tests. I never print from SD, so it seems we're uncovering all the places I don't normally look when testing :) thanks again for helping shake out these bugs so we can have a clean release.

smartin015, 2022-07-20:
Okay - I think I've fixed the SD printing behavior in [2.1.0rc3](https://github.com/smartin015/continuousprint/releases/tag/2.1.0rc3). My printers are acting up and not initializing the SD for OctoPrint to view, so I tested using a virtual SD card config option provided by OctoPrint. 

@winchester1094 please give this a try, and provide a sysinfo bundle if you still encounter problems. Thanks!

Also, sadly there are a bunch of new features that just fundamentally won't work with SD files (path validation, saving as .gjob, LAN printing). This is because many printer firmwares lack a gcode command for reading files back from the SD card (see [here](https://marlinfw.org/meta/gcode/), M20-M30 has functions for managing SD state, writing files, and listing files, but no "read" commands). For these cases I've added validation checks and visible errors so that the user at least knows the issue is due to a file being on SD and not on the OctoPrint host disk.

smartin015, 2022-08-03:
Hey @winchester1094, @napter - I'm back. I plan to promote the latest release candidate and close out this bug since it's soaked for two weeks and I haven't heard any further issues. Please ping this bug or open a new one if there's still problems, and I'll happily take a look. Thanks!

closed 2022-08-03 by smartin015