smartin015/continuousprint #92: Behavioral edge cases

smartin015, 2022-07-07:
**Describe the bug**

There are several non-deal behaviors that should be changed:

* [ ] Manually cancelling the print **from the printer's interface** is not treated as a user cancellation and the queue retries the print. This should instead pause the queue.
* [ ] Dragging and dropping a `.gjob` file with auto-add behavior on currently adds it as if it were a `.gcode` file - which is to say, erroneously. It should unpack the job file first before adding.
* [ ] There have been anecdotal reports of print files being run twice when they should only be run once
* [ ] The queue doesn't catch when the user increments the count of a job that's been fully completed (error in double-counting end state)

cheftip, 2022-07-07:
I am using the continuous print plugin for about a month now. 
I have had the problem with prints running twice when they should only be run once several times. Recorded it a few seconds ago. The bar isn't changing from blue to green. The counter doesn't work. Print History plugin just saved it as successful print.

Would be amazing to fix this bug! Other than that is the plugin awesome. Thank you!

https://user-images.githubusercontent.com/108418143/177871461-ad4d81af-0fd1-44e7-9239-f462d3a58fbb.mp4

[image]

smartin015, 2022-07-07:
Hey @cheftip, thanks for the report - can you upgrade the plugin to the latest version? It should be `2.0.0`, but from your picture it looks like you're running `1.6.0` or earlier. Just want to make sure I'm not chasing bugs in an older release, thanks!

Also the next time you see this behavior, can you please [attach a sysinfo bundle](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887)? The bundle has logs that can help with troubleshooting.

smartin015, 2022-07-12:
Ok, splitting some of these into different bugs and submitted fixes for the remainder. I'll be closing this bug out as it's overly general - leftover problems should be submitted as new issues.

closed 2022-07-12 by smartin015