smartin015/continuousprint #99: Improve job count interface

napter, 2022-07-15:
Currently when entering a job count, the text box displays how many items in total you want to print. After some items print this becomes a very non-intuitive number to manage for a few reasons: If 23 items items complete and you find 13 failed so have to increase the total to 36 - which doesn't seem right because what an operator cares about is how many are going to be printing. Also this leads to really strange behavior when reducing the number. If the total says "23" and I adjust it to 24 then it will say 23/24. But if I adjust it to 15 it will change to 15/15. 

I think this is a much cleaner way:
The total prints, and the number already printed should not be editable - those are calculated. The text box should be where you can edit how many are remaining to print. So if you set 23 to print and 20 already printed then it will say - completed: 20. total: 23, remaining 3. The remaining is what I edit - so if I want it to be done after the current one I change it to 0. If I want it to print 5 more instead of 3 more I change it to a 5. If it completed it would say: completed: 23, total: 23, remaining 0. If I now want to print 3 more I change the zero to a 3. Easy and intuitive and no funny shenanigans where I can mess with how many already printed (total is always calculated as completed + remaining).

napter, 2022-07-15:
I also noticed I can't edit the count of job that is printing while it it is printing - I am not sure the reason but if makes sense to me to be able to change how many are remaining even while printing (perhaps the one that is actively printing only lets you reduce it to 1 rather than 0).

smartin015, 2022-07-16:
> The text box should be where you can edit how many are remaining to print.

This is really good feedback - thanks for thinking about it in depth. I agree that editing the remainder is way more intuitive to work with, especially with larger counts. 

I'd like to bounce a couple ideas around, if you're open to a little discussion:

In the case of saving jobs as a `.gjob`, currently the counts of the job and inner sets are recorded. In this case, I imagine two possibilities:

1. The remainder gets saved - completed jobs are then saved with `remainder=0` for everything, and you have to set it yourself (perhaps by resetting the job's progress before saving).
2. The count still gets saved - IMO this requires the total count to be editable, otherwise it'd be frustrating to add back to the queue with the correct number.

For resetting job progress (i.e. select job, click the circular arrow button), For a job with `completed: 23, total: 23, remaining: 0`, I expect it would change to `completed: 0, total: 23, remaining: 23`. At this point, you could affect the "total" because everything is yet to be printed.

> makes sense to me to be able to change how many are remaining even while printing

It was convenient during the json->sqlite overhaul to treat jobs as "locked" once printing has begun. However, I do see the benefit of being able to edit more details of the job that do not affect the current print. Your suggestion of editing the remaining prints actually makes this a lot easier conceptually, but it's still a nontrivial amount of work. I've split this off into #101 so as to keep this issue focused on the user input parts.

smartin015, 2022-08-12:
Playing with this some more, I'm having trouble reconciling "just edit remaining" with the sets within a job. For instance:

```
Job 1: completed: 3, remaining: 2, total: 6/10
  - Set 1: completed: 1, remaining: 1, total: 6/10
```

If I edit this job by adding 1 to the set's remaining value as described above, I get:

```
Job 1: completed: 3, remaining: 2, total: 6/12
  - Set 1: completed: 1, remaining: 2, total: 15
```

...which is to say, the set's total also increases by 1 to account for the static completion amount and the increase in remaining.  But this actually increases the total number of `Set 1` printed by 2, as `Job 1` has 2 remaining. Worse, the user can't decrease the total and undo this potential mistake if the set's remainder has been consumed.  

That leads me to believe that we do need to adjust the count of sets within the job, which required a bit more work to implement. (draft in #114).

@napter, PTAL at the gif below and let me know if this works for your use case:
[Screen recording 2022-08-12 10.06.17.webm](https://user-images.githubusercontent.com/607666/184371249-94ada3ef-43fe-4008-8cca-03233c9859e7.webm)

smartin015, 2022-08-20:
Hey @napter - did you want to weigh in on these changes?

smartin015, 2022-10-15:
Closing this out as a new remainder-based UX is now publicly available on [v2.2.0rc1](https://github.com/smartin015/continuousprint/releases/tag/2.2.0rc1) and will become a full release after a couple weeks to soak.

closed 2022-10-15 by smartin015