smartin015/continuousprint #104: Clearing script Based on the job / Clear or pause

Kagnaw, 2022-07-23:
**Is your feature request related to a problem? Please describe.**
Based on job, i need to do different things. Some jobs i want to wait for user input/manual clear.
Some Prints needs to be cleared differently.
Also a print time saver don't have to clear the hole bed just because.

**Describe the solution you'd like**
Possibility to chose a clearing script on job basis.
or maybe a option to use bed clear or pause instead.

**Describe alternatives you've considered**
Currently have separate profiles when slicing. Adding bed clearing in the End G-Code.
honestly not that much of a hassle but this needs toughs before slicing and i sometimes are not that great at thinking.

**Additional context**
Thank you for taking your time reading.

smartin015, 2022-08-03:
Hey @Kagnaw, thanks for the request. Could you share what printer you're using, plus an example for each of

* a job where you want to wait for manual input
* a print that has to be cleared differently (and how)
* cases where you don't have to clear the whole bed

I'm happy to add more intelligent bed clearing, but I really want to look for solutions which don't require a lot of manual configuration and can instead be done automatically based on some kind of data/signal (e.g. the size of the print)

3DCoded, 2022-09-06:
This would be useful for prints with large surface area should cool down the bed first, and prints with small surface area doesn't have to wait to clear.

mlebaugh, 2022-09-26:
I came here to suggest a similar thing, I'm printing some objects that are larger than my clearing ability, would be nice if we could specify bed dimensions and required space for automatic clearing. If the object exceeds the clearance, trigger a wait for user instead of the clearing script.

smartin015, 2022-11-15:
Brief status update - I've been [refactoring](#148) the script automation side of CPQ with "conditional" events in mind. Once scripts can be attached arbitrarily to events (such as "queue print has finished"), I plan to follow up by giving users the option to conditionally run their scripts based on information other than the event itself. 

I expect it'll be most often used for different clearing behaviors based on print size, but the design will be general enough that other conditionals could be applied - to e.g. sweep the print off to the left or to the right depending on the material it was printed with, or to require manual removal any time a print file has `_manual` in its name. The main limitation will be in what's exposed for the conditional to evaluate upon. I intend to use [ASTEVAL](https://newville.github.io/asteval/) to provide a safe-ish evaluator that folks can "just use python" to configure, without having to pull in external scripts.

smartin015, 2022-11-16:
Okay, conditional execution is merged and should be ready to use in the next release (`v2.3.0`). The release will include updates to docs describing how to set up conditions, and I'll probably also do another release video.

closed 2022-11-16 by smartin015