smartin015/continuousprint #108: Continuous Print breaks file analysis

lettore, 2022-08-07:
I passed the last day trying to find out why when I upload a file to Octoprint I get "Model contains no extrusion" in info as Octoprint filemanager report octoprint.filemanager.analysis - INFO - Got output: ''.
After I disabled your plugin file analysis worked again, so I reverted back to version 2.0.0 and that fixed the issue.
So version 2.1.0 is bugged but I can't say where because nothing appears in the log.

Simonlebucheron, 2022-08-08:
I can confirm that, since I have updated C.P. to 2.1.0, [PrintTimeGenius plugin (2.2.8)](https://github.com/eyal0/OctoPrint-PrintTimeGenius) does not work and [Filament Manager (1.9.1)](https://github.com/OllisGit/OctoPrint-FilamentManager) is bugged.

smartin015, 2022-08-08:
Thanks for the reports with details - and sorry this wasn't a clean upgrade for your setups. I'm glad a revert mitigates things at least. 

I'll look into compatibility with the two plugins you mentioned, and with file upload - it'd help to have more details though:

@lettore: Are you uploading via the webpage (i.e. drag/drop or clicking the Upload button)? Can you attach an example gcode file that causes the message?

@Simonlebucheron: Can you provide more details beyond "does not work" and "is bugged"? I can install these plugins in a test environment and mess around with them, but it'd help to know exactly what behavior you saw that was problematic.

Simonlebucheron, 2022-08-08:
`[PrintTimeGenius plugin (2.2.8)](https://github.com/eyal0/OctoPrint-PrintTimeGenius) does not work`: the function of this plugin is to analyse Gcode after upload to octoprint to estimate correct time, and this does not work, even if I manualy analyse gcode in the plugin settings.

`[Filament Manager (1.9.1)](https://github.com/OllisGit/OctoPrint-FilamentManager) is bugged`:the function of this plugin is to manage spool and count remaining filament, but I do not know if it analyses gcode after uploading or if it subtracts length of filament on every extrusion. I noticed it was bugged because I have an option enable which remind me to select the correct spool after I click on "print", a window pop up as usual but I could not select the spool, like there were no spool saved in the plugin database (which were well saved when I looked in the settings). The print cannot start so I had to unselect this option and I could continue my print. I think the plugin was working fine after that (beside this pop up window), but I did not check the evolution of my spool filament during this time.

I hope it helps ;)

smartin015, 2022-08-08:
So - `v2.1.0` added some basic file analysis that reads the headers of gcode files in an attempt to parse metadata for fancy automation (e.g. automatic printer selection over network queues). This analysis is independent of the default analysis which occurs on file upload and provides extrusion and other metadata based on the gcode commands in the file.

However, it looks like my assumption of how file analysis works was wrong - registering a new analyzer doesn't append it to a list of analyzers to run when a file is added - it **totally replaces any analyzer registered for that file type**. That's preventing the "regular" analysis from completing and providing extrusion estimates (likely affecting FilamentManager) and it replaces the registered analyzer provided by PrintTimeGenius (which is also doing the same analysis replacement). 

Ideas:

* rely on the [callback system](https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/filemanager/analysis.py#L89) and treat completion of "regular" analysis as an event to start CPQ analysis. I don't know if that'll always complete properly, though.
* hook into [process_backlog](https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/server/__init__.py#L1078) - but that only [queues items to the main analysis queue](https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/filemanager/__init__.py#L324), and only on startup (AFAICT).
* Chain analysis scripts by getting and calling the previously registered analysis script when run. This only works if CPQ is the last analysis plugin to initialize, otherwise it'll get replaced by subsequent plugins and fail to analyze.
* Roll my own analysis queue, using [set_additional_metadata()](https://docs.octoprint.org/en/master/modules/filemanager.html#octoprint.filemanager.storage.LocalFileStorage.set_additional_metadata) to store metadata. Honestly this doesn't seem like that much work at this point, and it eliminates basically all compatibility issues.

To DIY as per the last bullet, I'd need an equivalent work-generator to `process_backlog` - read all file metadata, find ones missing analysis, and enqueue them for processing. Then I can create a separate `AnalysisQueue` instance and run it myself, adding a callback which  calls [set_additional_metadata](https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/filemanager/__init__.py#L1055) using my own namespace (i.e. not `analysis` which is the default). Also need to listen for `FILE_ADDED` events and enqueue new analysis as needed. This'll be my initial approach since it's most likely to fix the problem Once And For All. 

Assuming that works as a fix, I'll push a `2.1.1` patch to restore compatibility and fix broken analysis.

smartin015, 2022-08-10:
Okay - released `2.1.1rc1`, which separates analysis from the default analysis queue and should fix compatibility. 

@lettore, @Simonlebucheron: can you both enable release candidates [per the instructions towards the bottom of this page](https://smartin015.github.io/continuousprint/), then update to the RC and let me know if that fixes things for you? Thanks!

Simonlebucheron, 2022-08-10:
That is quick support, thanks ;)
I am currently prototyping my machine, I will try that when I return to a nominal configuration.

Simonlebucheron, 2022-08-14:
> Okay - released `2.1.1rc1`, which separates analysis from the default analysis queue and should fix compatibility.
>
> @lettore, @Simonlebucheron: can you both enable release candidates [per the [...]

It is working for me ! At least the two plugins and functions quoted previously. I did not test yet the continuous print main functions but it looks good. Thanks !

smartin015, 2022-08-20:
That's great to hear @Simonlebucheron! I'll let this soak a couple more days to give other folks a chance to report any incompatibilities, then do a full release to bring everyone else up to speed.

smartin015, 2022-08-24:
Okay - it's now a proper release. Please open a new issue if you encounter any more problems - thanks again!

PR #112 (2022-08-10): Analysis & concurrency fixes
May help mitigate #109

Should fix #108

Improves logging state transition to clearing/finishing, which should help understand issues where the queue over/under-prints a job.

Testing:
* [x] Analysis backlog generation picks files without analysis and excludes those with analysis
* [x] Newly added files are queued for analysis
* [x] Analysis completion updates additional metadata and fires event
closed 2022-08-24 by smartin015