smartin015/continuousprint #109: Continuous print don't count correctly the number of print.

jeremy-3dproject, 2022-08-08:
**Describe the bug**
Hello, First of all thanks for this great plugin !

I had some issue on the plugin and don't know if this happens to others peoples.

When I ask the plugin to do a new job of 6 times the same print job with only one file, the plugin execute the job and clear the bed 5 times and not 6 times. 

I try to update to the last version, delete the job and try with other number and every time I had one print less compare to what I set.

**To Reproduce**

Don't know if this happen's to other people. 
If needed, we can try it with a teamviewer session.

**Expected behavior**

Print the same number of job then I set. 

**Screenshots**

[image]

**System info:**
Octoprint 1.8.1 => System info will be uploaded if needed.

**Browser logs:**
The printer is actually working on a print job, I will let it finished and collect the log file to if there is something strange.

Thanks for your help and have a nice day !

Best Regards,

Jeremy

smartin015, 2022-08-09:
Hey @jeremy-3dproject, thanks for reporting this. To help me root-cause, I'll need you to do the following:

1. Configure and start a job that prints your file with the file itself set to `count=1` and the job set to `count=2`
2. Send me a sysinfo bundle after you observe only a single copy printed.

I'll take a look at the logs and hopefully find a pattern as to what's causing this.

FWIW I tried the same thing with a virtual printer, and it actually printed one time too many, but subsequent jobs behaved as expected. That actually led me to a thread-safety issue with state handling, which I'll fix and publish in `v2.1.1`. 

Not sure if that's 100% related to this, though. Something's definitely fishy - I'm adding additional debug logging in the next release to make it easier to see where the logic is breaking down.

jeremy-3dproject, 2022-08-10:
Hello @smartin015 thanks for reply me so fast ! I really appreciate it :)

The printer have actually finished running with a 5 count print job (see screenshot) and print only 4 times and not 5.

I have uploaded the sysinfo bundle and oppefully you can found something into the log. 

If its easer for you to have only a 2 count print like asked, just tell me and I will do it. 

Thanks for your help and have a nice day
[octoprint-systeminfo-20220810055723.zip](https://github.com/smartin015/continuousprint/files/9296534/octoprint-systeminfo-20220810055723.zip)
[image]
[image]
.

smartin015, 2022-08-10:
Thanks for your sysinfo bundle! I do see some evidence there of what I saw in my own test setup: the state machine that drives queue behavior was having issues with multiple threads trying to change its state at the same time. 

I've released [2.1.1rc1](https://github.com/smartin015/continuousprint/releases/tag/2.1.1rc1) which includes a couple tweaks aimed at fixing this. 

Please enable release candidates as per the bottom of [this page](https://smartin015.github.io/continuousprint/), and let me know if that fixes things or if you still see the same problem.

smartin015, 2022-08-20:
Hey @jeremy-3dproject - have you had a chance to try the new RC?

jeremy-3dproject, 2022-08-20:
Hi Scott,

Sorry for my late feedback.

Yes I try it few days ago and for the moment it's seems that it work 😉.

I keep you updated by the end of the next week.

I will try to change the number of print to see if it work all the time.

Thanks and have a nice evening.

Best Regards

Jeremy

smartin015, 2022-08-24:
I've promoted the changes to a full release (`2.1.1`) so I'll close out this issue. Please open a new one if you see more problems like this - thanks again!

closed 2022-08-24 by smartin015