smartin015/continuousprint #120: randomly skipping file in job

TylonHH, 2022-09-05:
**Describe the bug**
So far the plugin worked fine, but suddenly it skips files. I have just one job with two files. I noticed that one file is only printed. I expected that both files are order one by one. Ressting the job didnt help. I have this issue on two different printers (same file).

**To Reproduce**
I donno

**Expected behavior**
All files are printed.

**Screenshots**
[image]

[expected octoprint log](https://pastebin.com/a2FZR29i)
[failured log](https://pastebin.com/vyRs87vh) only bottom printed
Hint: _E_V3_newest_BOTTOM_vase_mode.aw.gcode_ and _E_V3_newest_BOTTOM_vase_mode.aw.gcode_ are the files. 

**System info:**
[octoprint-systeminfo-20220905231119.zip](https://github.com/smartin015/continuousprint/files/9491996/octoprint-systeminfo-20220905231119.zip)

**Browser logs:**
Collect the console logs for your browser and upload them to this issue. See [here](https://support.shortpoint.com/support/solutions/articles/1000222881-save-browser-console-file) for instructions on how to do this for different browsers.

**Additional context**
I see a failure in the log, but no reason. you?

TylonHH, 2022-09-09:
it happen again. Why is [this print](https://pastebin.com/xAAmV4e6) aborted? It was printed fine, the bed clearing also finished, and the file (bottom) was printed again. But why?

smartin015, 2022-09-10:
Hey @TylonHH - thanks for the bug report! I finally had a moment to look at the problem this morning, and I think it's related to a small behavior change I made in `v2.1.1` which assumed the `PRINT_DONE` event would always fire when the print completes.

For some reason, your octoprint instance is not sending the `PRINT_DONE` event to the CPQ plugin when a print finishes. I don't know if this is a plugin compatibility issue or whether Octoprint just has flaky event handling. 

Either way the behavior is wrong, so I'm releasing `v2.1.2` to RC as an attempted fix. I released it in a weird way, so I don't know if the plugin updater will actually see it. If that happens, try installing it per step 3 of [this page](https://smartin015.github.io/continuousprint/getting-started/). 

Let me know if this solves the abort issues for you, and I can do a full release.

TylonHH, 2022-09-14:
So far since six jobs without an error.

closed 2022-09-26 by TylonHH