smartin015/continuousprint #122: Add Owner of print job if missing

lettore, 2022-09-14:
Hello,
I use the plugin Print Job History to track my print history, when I use your plugin to manage the print queue I see as the owner of the print job "John Doe".
This happens because Print Job History can't find the owner of the printed job.
https://github.com/OllisGit/OctoPrint-PrintJobHistory/blob/eb49a9547b7b885b9d3f46f9df37b9b582e08739/octoprint_PrintJobHistory/__init__.py?plain=1#L18
It's possible to add this value as the user that made the queue?
Thanks,

smartin015, 2022-09-19:
Hey Lettore - thanks for the request, and for the context! It's refreshing to get a link to reference code :)

I agree with this change. Interestingly, the [API docs](https://docs.octoprint.org/en/master/modules/printer.html#octoprint.printer.PrinterInterface.start_print) fail to mention there's an optional [user param](https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/printer/standard.py#L727) when calling `start_print()` that would set the user. Should be easy enough to just [grab and use the current user](https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/server/api/job.py#L29).

smartin015, 2022-10-15:
Fix has been made and staged in branch `rc2` for release after `v2.2.0rc1` is promoted to full release.

smartin015, 2022-11-11:
`rc2` now merged into `rc` and will go out on next release. I'm closing this out as there's no work left to do.

closed 2022-11-11 by smartin015