smartin015/continuousprint #124: @pause in bed clearing script does not work

Tarinaky, 2022-10-04:
**Describe the bug**
Using @pause as the bed clearing script does not work - printer immediately continues printing.

**To Reproduce**
Steps to reproduce the behavior:
Set the bed clearing script to @pause
Add two gcode files to the print queue.
Run the queue / Click start managing

**Expected behavior**
Printer should pause when the first print is completed - so that a human can reset the build plate manually.

smartin015, 2022-10-04:
Hi @Tarinaky - thanks for your report. `@pause` commands have been pretty thoroughly tested at this point, so I'm surprised you're seeing trouble here. 

Can you please fill out the rest of the issue template, especially [generating a sysinfo bundle](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) so that I can look at the logs, check for plugin conflicts, get version info, etc?

smartin015, 2022-10-14:
Hey @Tarinaky, are you still having issues? If so, could you please grab that sysinfo bundle? Thanks.

Tarinaky, 2022-10-18:
Hi. Sorry, I've been struggling to find the time to get back to you on this and haven't had the time to get the sysinfo bundle but wanted to share some later debugging I did:

I queued up some empty gcode files to test this, but I think Continuous Print is running the 'before each print' gcode script (I can't remember the exact setting name in octopi, appologies) and inserted into the start of the bed clearing script, which in my case is where I put my autobed levelling sequence. It then pauses after performing that bed levelling - but in practice this causes a print head crash for me.

Is there any way to toggle this behavior perhaps, so that the bed clearing script isn't having that global gcode inserted?

smartin015, 2022-10-19:
I see, thanks for digging into the problem. It makes sense that 'before each print' is triggering, as from Octoprint's perspective the bed clearing script is a "print". I can see where this is undesirable, but AFAIK there's no way to change this behavior without a major overhaul (and probably modifying Octoprint itself).

One way you could address this without any plugin changes is by putting your auto leveling sequence into your gcode directly - for whatever slicer you use, there's usually a "gcode header" setting where you can paste in some custom code before the body of the file. That way, your autoleveling would only happen when your gcode file is run, and not before the clearing/finishing scripts.

It's a bit brittle - if you change your slicer settings, the autoleveling might get accidentally removed. I could imagine a "before each queue item" equivalent in CPQ settings which functions similarly to octoprint's "before each print", just only with CPQ items. I'll add that onto #126 as it's in the category of "run gcode when <event> happens", and needs a bit of refactoring before it can be done well.

LMK if the header workaround is sufficient, and we can close this out in favor of tracking event-scripting in #126. Thanks!

smartin015, 2022-11-11:
I'm going to close this out since discussion has stalled; will work on event-scripting suggestion in #126.

closed 2022-11-11 by smartin015