smartin015/continuousprint #126: Filament change

Faultier523, 2022-10-13:
It would be cool to add a possibility for a script when a filament change is required, for example to automatically move to a position and eject the filament.

smartin015, 2022-10-14:
Agreed, that would be cool. I think I'll expose this after I've refactored script-running to be a bit more general, though... currently each script/event requires code and tests to be added in several places before they work. That's OK when there's only a couple scripts, but it seems like "on queue event, run script" is a recurring feature request :)

smartin015, 2022-11-16:
With submission of #149, these events will be available to run scripts on in `v2.3.0` - release ETA is probably some time in December.

PR #149 (2022-11-16): Add events & scripts for activate/deactivate, filament swap
Closes #126.

* [x] Test and also verify that ACTIVATE and PRINT_START scripts run to completion sequentially
* [x] Test that DEACTIVATE, PRINT_START, and AWAITING_MATERIAL scripts are correctly executed
* [x] Verify that all events are firing correctly in virtual environment.
closed 2022-11-16 by smartin015