smartin015/continuousprint #152: LAN Jobs printer information to be more clear

simoncrabbuk, 2022-11-16:
Currently have to hover over cube to find out which machine a job is running on.

Would like an actual printer name always displayed on the active LAN jobs.

Also, on the printer that's running the job need to make it more clear which job it's running, currently they all look the same.

smartin015, 2022-11-16:
Thanks for the request! I'm actually looking for feedback on what form this should take, which is why it's a bit basic at the moment. 

Maybe having the printer name visible would be enough distiction... or there could be different icons / color emphasis to indicate local vs lan. Do you have any specific suggestions for distinguishing local vs lan printers running jobs?

simoncrabbuk, 2022-11-17:
I'm thinking if the printer name running the job is displayed that will help a lot.

Jobs running locally should perhaps be in the current blue green colours, and jobs running on other printers should be pale versions of the same.

Though, could I suggest Red Amber Green for preference of colours!  Red = failed, Amber = running, Green = done.