smartin015/continuousprint #154: Continuous Print API Error 500

Dom24123, 2022-11-16:
Continuous Print API Error 500 when clicking "start managing button" for continuous print plugin, G code file is added on the job queue.

This is not as expected, since the printer should start loading the job and print.

[image]

[octoprint-systeminfo-20221116204757.zip](https://github.com/smartin015/continuousprint/files/10024806/octoprint-systeminfo-20221116204757.zip)

[image]

smartin015, 2022-11-16:
Thanks for the sysinfo bundle - I just took a look at the logs and it appears to be an issue with port discovery:

```
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\octoprint\plugin\__init__.py", line 273, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "c:\octoprint\venv\lib\site-packages\octoprint\util\__init__.py", line 1688, in wrapper
    return f(*args, **kwargs)
  File "c:\octoprint\venv\lib\site-packages\continuousprint\__init__.py", line 64, in on_after_startup
    self._plugin.start()
  File "c:\octoprint\venv\lib\site-packages\continuousprint\plugin.py", line 82, in start
    self._init_queues()
  File "c:\octoprint\venv\lib\site-packages\continuousprint\plugin.py", line 289, in _init_queues
    lq.connect()
  File "c:\octoprint\venv\lib\site-packages\continuousprint\queues\lan.py", line 39, in connect
    self.lan.connect()
  File "c:\octoprint\venv\lib\site-packages\peerprint\lan_queue.py", line 158, in connect
    self.spin_async()
  File "c:\octoprint\venv\lib\site-packages\peerprint\discovery.py", line 63, in spin_async
    self.sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEPORT, 1)
AttributeError: module 'socket' has no attribute 'SO_REUSEPORT'
```

Apparently `SO_REUSEPORT` exists on Linux, but not on Windows - more detail [here](https://stackoverflow.com/questions/14388706/how-do-so-reuseaddr-and-so-reuseport-differ?rq=1). As I mainly use Linux, I didn't notice this compatibility issue. I'll start on a fix, but it'll take a little bit to publish due to some other changes in the pipe. 

In the meantime - if you can get to the Settings page, try going to `Continuous Print > Queues` and removing the entry marked `LAN`, then restarting OctoPrint. This should remove the initial connection logic and make it work locally for you at least.

Dom24123, 2022-11-17:
Thank you Scott

This has gotten the continuous print working, you are a legend amongst men!

Where can i get bed clear G code for creality Ender 5, to use the extruder
head to knock the print off the bed? i go not fin any example codes in the
gcode script section?

Kind Regards
Dominic Truter
072 981 6654

smartin015, 2022-11-27:
`v2.3.0` will surface this error but make it so you can at least use the queue - I'm leaving the bug open until I fix the root cause.