smartin015/continuousprint #157: Not counting printed parts correctly.

Tony-321, 2022-11-18:
The Plug in is not registering printed items and therefore not counting them correctly. No idea why.

It’s stating 3/15 items printed and printing No 4, while it’s actually printed 5 already and it’s printing the 6th, it did the same thing on the previous item in the que as well. Printed 3 more than I needed.

As you can see in the picture in the box it’s the square piece on the left.

[image]

it’s saying 
[image]

smartin015, 2022-11-19:
Hey @Tony-321 - thanks for the bug report. 

Can you please attach a sysinfo bundle so I can take a look at the plugin logs?

smartin015, 2022-12-16:
@Tony-321 - I'm closing out this issue as I can't do anything for you without a sysinfo bundle to understand what's going wrong. If you find the time to submit one to this issue, I'll reopen it again to see what's up.

closed 2022-12-16 by smartin015

Tony-321, 2022-12-16:
Hi mate, no problem.I upgraded my RaspberryPi and it seems to have resolved the issue.Perhaps an old PI can’t run the application correctly.Tony McCormackMHE Fleet Engineer.Mobile : 07841404332