smartin015/continuousprint #161: Clearing bed script don't do all gcode command

Theskully, 2022-11-22:
**Describe the bug**
Hi,
Clearing bed script don't do all gcode command, my clearing bed script :
OCTO2 (my gcode command to stop my fan cooling my bed)
g4 s10
g0 x125 y210 z1
g0 y0
g0 x0 y0 z10

**To Reproduce**
It happens on several of my printers not all the time sometimes randomly, i can't figure out why....

**Expected behavior**
My printer execute my clearing bed script entirely

**System info:**
[octoprint-systeminfo-20221122230919.zip](https://github.com/smartin015/continuousprint/files/10071722/octoprint-systeminfo-20221122230919.zip)

**Additional context**
I have Prusa MK3S or Ender 3 S1

smartin015, 2022-11-23:
Hey @Theskully, thanks for the report and for the sysinfo file. I took a look at your logs and I don't see any related errors coming from CPQ. 

If I had to guess, I think you may have set up some conflicting automation between CPQ and [BedCooldown](https://plugins.octoprint.org/plugins/bedcooldown/). For instance, I see things like:

```
2022-11-22 21:07:07,557 - octoprint.plugins.bedcooldown - WARNING - _bedcooldown_timer_triggered triggered but not printing? This shouldn't happen.
```

Try disabling the BedCooldown plugin, and instead use the [managed bed cooldown feature of CPQ](https://smartin015.github.io/continuousprint/managed-bed-cooldown/) if that works for your setup. Let me know if this fixes things, or if you still have problems.

Theskully, 2022-11-24:
I think thats it, i'll test it on my farm, it seems work (disable plugin BedCooldown) but my bug was not always happening, i'm not really sure for the moment.

smartin015, 2022-12-08:
Hey @Theskully, I'm going to close out this issue since it sounds like you found a fix. Please leave a comment if you're still having problems and we can open it up again. Thanks!

closed 2022-12-08 by smartin015