smartin015/continuousprint #167: LAN queue deleted if system turned off.

simoncrabbuk, 2022-12-07:
I keep losing my LAN queues, I turn my PIs off remotely, without shutting them down, and most of the time there's no problems.  But I'm finding my LAN queus are not there when I power back up, and have to reconfigure them.

2.3.0-rc1

smartin015, 2022-12-08:
Thanks for the report - can you clarify whether it's the LAN queue itself that's deleted, or its contents? 

If it's the queue itself (i.e. missing from Settings) that's really unexpected. 

But if it's the contents, that's "working as implemented" as persistent journaling is turned off at the moment. It's disabled because I kept seeing the queue stall out when trying to fast-forward from saved logs... something I hope to have fixed in the next iteration of [PeerPrint](https://github.com/smartin015/peerprint), which I'll be integrating after I push out [auto-slicing of STLs in the queue](https://github.com/smartin015/continuousprint/pull/166).

simoncrabbuk, 2022-12-22:
The queue itself disappears!  Have to go into settings and Add Queue on every machine...  But, it doesn't ALWAYS do that!   I've tried turning various machines off and on, ALL machines off and they're all coming back with the queue present.  I'll try and gewt a dump when it happens again.

smartin015, 2022-12-22:
Thanks - I'm interested to see if there's any errors in saving or startup that make the queue seem to not be visible... queue configuration uses the same storage system as all other octoprint settings, so I'm surprised that it would be disappearing even with a hard power-off if the same problem isn't seen elsewhere.