smartin015/continuousprint #168: Unable to delete files on Octoprint 1.8.6

lettore, 2022-12-11:
Unable to delete files stored in Octoprint version 1.8.6 with Continuos Print 2.2.0.
It's not the first time this happens, I kindly ask if there's a way to not interfere with Octoprint basic functionality as looks like every minor Octoprint update breaks this function.

simoncrabbuk, 2022-12-11:
I've seen this behaviour in 2.2.0, but I've been testing 2.3.0rc1 and it seems to be resolved.

lettore, 2022-12-11:
Yes thank you, but it's quite frustrating if you work on production systems to have broken functionality and to have to disable on plugin at time to find the culprit. I opened a ticket in Octoprint Github to see if there's a better way to handle such problems, a plugin can't disrupt basic functionality.

smartin015, 2022-12-11:
Hey @lettore - thanks for your report and the feedback. I feel your pain here, as the release process of OctoPrint and its plugin ecosystem are somewhat disconnected (AFAICT this is by design, to limit the burden of the core maintainers).

I absolutely welcome your ideas to catch bugs earlier in the release process. 

To share what I already have in place:

1. I soak every release candidate in the `rc` branch for a minimum of two weeks before pushing to production. This gives production-minded users a chance to try the RC on a non-critical setup and report bugs, which I consider as blockers to a full release.
2. I manually [QA check](https://github.com/smartin015/continuousprint/pull/123) more change-heavy release candidates before release, to catch any basic issues with CPQ behavior. This is done on a virtual instance of octoprint with a couple of plugins I know to be commonly used by folks (e.g. SpoolManager). 
3. Coming out in the next release are some [improvements to the bug reporting flow](https://github.com/smartin015/continuousprint/issues/163) that make it easier for folks to find this github issue tracker and report bugs.

I have a couple known blind spots, as well:

1. I don't know definitively what plugins folks are using. A release that works fine on a clean install of OctoPrint may error if another plugin happens to have conflicting behavior (e.g. #161).
2.  I don't know for sure how folks are using my plugin. This runs afoul of [Hyrum's law](https://www.hyrumslaw.com/).
3. I am heavily reliant on users reporting their issues to me. If I don't hear of a user issue, I can't know that it exists unless it happens on my own setup or in development.

To shore up my own release process, I can think of a couple improvements:

1. Explicitly update the octoprint instance to the latest version before doing QA
2. Also QA "basic" octoprint behaviors (e.g. file drag-drop, upload button, deletion, printing through vanilla flow)

I know basically nothing about your setup (see blind spots 1 & 2) - but some generic advice to improve your own production setup if you haven't done so already: 

You could set up a staging environment of one or more printers (virtual or otherwise) with all installed plugins set to `rc` branches, and updating whenever a new release emerges. This environment should run similar workloads as you do in your prod printer farm, but make exceptions highly visible so they can be reported early. This would catch behavioral bugs not just for CPQ, but for any and all plugins you have installed in your prod setup - before they impact the running of your print farm.

**TL;DR: I'm trying to provide a free, reasonably stable plugin that operates in an unstable ecosystem, and there may be things both of us can do to improve stability here.**

lettore, 2022-12-11:
Hi @smartin015 , I reported the problem as soon as I saw it, with no polemical intent, as I perfectly know that no developer can test every possible scenario.

As I said I opened a ticket in Octoprint Github but I got a not-really polite reply so I'm not going further there.

I'm used to managing at least 100 open source projects in my household, from Nextcloud, Home Assistant, Gitlab etc, for all of them I build my own Docker Image so I'm quite experienced in it.

I'm not a Python programmer but I know how to set up a CI, and when I build an image I usually do a test script that check if base functionality is working, for example, if a web server is responding correctly to a designed port.

I think you can do the same and build a Octoprint Docker image with installed your plugin and trigger a script that automatically uploads a file, loads it into a queue, renames it, deletes it etc.

This is to ensure that at least, basic functionality is not broken, then obviously everything can happen when you mix many plugins that alter how Octoprint behaves.

I feel that the main problem here is how much a plugin can go deep into Octoprint functionality, I already had many plugins that conflict with each other, and the only way to diagnose who is the culprit is to disable the plugins one by one.

I really appreciate your kind reply and your work.

I installed the release candidate version 2.3.0 rc1 and the problem is still there, this is the error in Chrome console
```
packed_plugins.js?7d548955:19868 Uncaught TypeError: Cannot read properties of null (reading 'jobs')
    at CPViewModel.self.files.removeFile (packed_plugins.js?7d548955:19868:47)
    at Object.eval (eval at parseBindingsString (packed_libs.js?7111420c:131:1741), <anonymous>:3:226)
    at HTMLDivElement.<anonymous> (packed_libs.js?7111420c:133:4368)
    at HTMLDivElement.dispatch (packed_libs.js?7111420c:13:43090)
    at v.handle (packed_libs.js?7111420c:13:41074)
```

smartin015, 2022-12-11:
> Hi @smartin015 , I reported the problem as soon as I saw it, with no polemical intent, as I perfectly know that no developer can test every possible scenario.

Acknowledged & understood - reading emotion through text is hard sometimes :)

> I think you can do the same and build a Octoprint Docker image with installed your plugin and trigger a script that automatically uploads a file, loads it into a queue, renames it, deletes it etc.

This is roughly where I'm at already - I develop using a [custom Docker image](https://github.com/smartin015/continuousprint/blob/master/Dockerfile) with my plugin installed. The [CI setup for CPQ](https://github.com/smartin015/continuousprint/blob/master/.travis.yml) doesn't use this image, but rather installs OctoPrint + CPQ to perform build & unit tests without some of the heavier stuff that's not needed for that use case. [Coverage is >90%](https://coveralls.io/github/smartin015/continuousprint?branch=rc), but of course that only accounts for CPQ behaviors.   

For integration testing as you suggest, I'd probably need to start setting up Selenium or similar tests. 

> I feel that the main problem here is how much a plugin can go deep into Octoprint functionality, I already had many plugins that conflict with each other, and the only way to diagnose who is the culprit is to disable the plugins one by one.

OctoPrint maintainers have their hands full with broader compatibility / stability and are probably (and understandably) leery of giving yet more more official/supported control over the myriad parts of their UI. That leaves plugin devs with the choice of either duplicating UI elements to stay within the "plugin tab" idiom, or of doing more brittle injection to provide a decent UX in spite of the guide rails OctoPrint has in place. 

Sometimes I toy with the idea of just having a separate webpage for CPQ and linking from the OctoPrint plugin tab... but that's a pretty large deviation from normal plugin behavior and a less cohesive experience. It'd be pretty cathartic, though. 

 > I really appreciate your kind reply and your work.

❤️ 

> I installed the release candidate version 2.3.0 rc1 and the problem is still there, this is the error in Chrome console
>
> ```
> packed_plugins.js?7d548955:19868 Uncaught TypeError: Cannot read [...]

Ah, unfortunate - opened #169 for tracking.

smartin015, 2022-12-28:
Hey @lettore - just a heads up that the mitigation (but sadly no solution, without a sysinfo bundle) for #169 has been soaking in `2.3.0rc2` for awhile. Please let me know if you have any further issues with it, otherwise I'll do a proper release in the next couple days.