smartin015/continuousprint #182: Internal Error 500 API

MadTerminator, 2023-01-01:
Hi, when I start the job the plugin get this error, how I fix it?

`2023-01-01 12:33:04,104 - octoprint - ERROR - Exception on /plugin/continuousprint/set_active [POST]
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/flask/app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/lib/python3.9/site-packages/flask/app.py", line 1519, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/lib/python3.9/site-packages/flask/app.py", line 1517, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/lib/python3.9/site-packages/flask/app.py", line 1503, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/lib/python3.9/site-packages/octoprint/server/util/flask.py", line 1575, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/octoprint/server/util/flask.py", line 1598, in decorated_view
    return func(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/continuousprint/api.py", line 86, in cpq_permission_wrapper
    return func(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/continuousprint/api.py", line 174, in set_active
    self._update(DA.ACTIVATE if active else DA.DEACTIVATE)
  File "/usr/lib/python3.9/site-packages/continuousprint/plugin.py", line 746, in _update
    if self.d.action(a, p, path, materials, bed_temp, timelapse_start_ts):
AttributeError: 'CPQPlugin' object has no attribute 'd'`

TylonHH, 2023-01-04:
*push* same here.
I updated the plugin and reconfigured my bed cleanin and finishing script. I also restarted OctoPrint.

TylonHH, 2023-01-04:
A clean reinstall of the plugin solved this

smartin015, 2023-01-04:
Hey @MadTerminator - that error is a generic catch-all for something going wrong on plugin startup, so I can't really offer help until you attach a system bundle (see [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887)). That lets me view all the logs. Thanks!

smartin015, 2023-01-17:
Hey @MadTerminator, are you still having this issue? I'll go ahead and close this out, will reopen if I hear back with a sysinfo bundle.

closed 2023-01-17 by smartin015