smartin015/continuousprint #183: Documentation update to Preprocessors

Terandium, 2023-01-03:
**Is your feature request related to a problem? Please describe.**
The current calling order of events is unclear, what happens if you have 2 events but only 1 preprocessor?

**Describe the solution you'd like**
Update the documentation to show what would happen in this case.

**Describe alternatives you've considered**
None, testing to wait and see XD

**Additional context**
[image]
In that screenshot will the "Pause Print" script run on every print success, or only the ones where "Bed Clearing" doesn't apply?

smartin015, 2023-01-04:
Hi @Terandium, thanks for trying the new preprocessors :D 

I'll update the documentation in a bit, but something that may help you in the meantime:

* An Event can have multiple Scripts; each Script is executed in the listed order when the Event fires.
* A Script can have either zero or one Preprocessor assigned.
   * If it does not have a Preprocessor, it is run as-is.
   * If it does have a Preprocessor, the Preprocessor is run first and its output is applied to that Script before that script is executed.
 
In your example, the `Pause Print` script will always be run as-is, no matter whether or not `Bed Clearing` has a preprocessor assigned. 

Note however that preprocessors can pass information to each other down the list (doc note [here](https://smartin015.github.io/continuousprint/gcode-scripting/#state-carryover)). If `Bed Clearing` had a preprocessor that set a variable `foo` before returning a value, that same `foo` variable would be available on a preprocessor for `Pause Print`.

Hope this helps!

Terandium, 2023-01-04:
Thanks a lot, that for sure clears things up. I would have found a way to make it work but haven't had the time to fiddle around with it yet.

Love the new preprocessors though, million of new possibilities!

smartin015, 2023-01-12:
Let's leave this one open so I remember to update the doc :)

smartin015, 2023-01-18:
Ok, added. I'll push the docs change out with the next full release.

closed 2023-01-18 by smartin015