smartin015/continuousprint #185: Spool Manager pause queue when out of filament

Terandium, 2023-01-05:
**Is your feature request related to a problem? Please describe.**
The plugin has integration with Spool Manager to use certain materials/colors, but it doesn't support filament tracking and thus it continues the queue when it runs out of filament.

**Describe the solution you'd like**
Pause the queue when it runs out of filament

**Describe alternatives you've considered**
Right now I am using preprocessors to do the same thing, I have checked the spool manager config but it only warns if it runs out of filament and doesn't seem to pause the print. (It does ask if you are sure when manually starting a print, but that doesn't really help with an automatic queue)

smartin015, 2023-01-05:
That's a great idea!

This'll need a bit of consideration though - for printers with a runout sensor (like my CR-30) you can run a print and change filaments midway when the last of the spool is gone. The printer pauses when the filament runs out, waits for it to be changed, and then picks right back up again where it left off. Adding extra behavior on top may actually be a worse experience in that case.

I'm guessing your printer doesn't have a runout sensor, so it just starts air-printing due to lack of filament - is that right? In this case, would you rather:

1. The printer doesn't attempt to print a file if it would run out of filament before the file is finished (similar to SM's warnings, but actively preventing printing at the start)
2. The printer waits until it's just about to run out (e.g. 10m of filament remaining) and then pauses mid-print so the user can change filament (more like the runout sensor behavior, but based on total filament extruded)

I'm personally a fan of \#2, but that actually sounds like a feature request to Spool Manager rather than CPQ.

Also, can you share the preprocessor you're using?

Terandium, 2023-01-11:
Sorry for the late answer as I didn't see your message ^^
I would personally go for option 1, as changing filament during print often leads to an ugly mark. (That is biased as I don't have a filament runout sensor and thus the entire experience is just bad haha, as the software solutions aren't as perfect as a physical runout sensor)

I can see why you'd pick option 2 and I agree that that should be a feature of Spool manager rather than CPQ. The problem then is that every spool has different amounts of filament, some are 1000kg while others are 1000.1kg so it could still print in the air. (or stop before, which is preferred then as it won't ruin the print)

The preprocessor doesn't actually do anything as I set up an MQTT server, and have a nonstop running Golang script that gets the remaining weight and posts it to the external endpoint.
But I ran into the issue of getting the metadata of the upcoming print, as the metadata kept being empty. On that note, maybe a preprocessor validator of sorts makes sense. Trial and error is quite a nuisance because I need to start and stop the queue to test events haha.

With all that I gave up and right now the bed-ready plugin checks if the print failed, because if it ran out of filament mid-print it won't be able to kick it off. and thus fail. For now fine I'll just have to get a filament runout sensor soon.

**TL;DR:** Getting a physical filament sensor is simply better than any software solution ever can be.
Maybe adding a validator or being able to manually run a preprocessor will be useful though, as it's difficult to currently test whether or not the preprocessor is working. (But that would have to be an issue on its own)

smartin015, 2023-01-12:
> ...I would personally go for option 1, as changing filament during print often leads to an ugly mark. (That is biased as I don't have a filament runout sensor and thus the entire experience is just bad haha, as the software solutions aren't as perfect as a physical runout sensor)

Totally reasonable - and having a runout sensor shouldn't be a requirement to use CPQ in any case. I think this is worth adding to the plugin, so I'll keep this issue open to do so.

> The preprocessor doesn't actually do anything as I set up an MQTT server, and have a nonstop running Golang script that gets the remaining weight and posts it to the external endpoint. But I ran [...]

Adding a validator was a lot of extra work for a feature I didn't know if people would use, so I figured I'd hold off on that until I collected a few opinions :) I just opened #190 - if you're up for it, I could use your input on where the sharp edges are so far and what would make preprocessors easier to work with!

Terandium, 2023-01-13:
I will check it out! I might even help out with some pull requests, but unsure as I am sadly quite busy and I haven't much experience with Python. I mostly work in Golang and Rust.

smartin015, 2023-03-08:
With https://github.com/smartin015/continuousprint/releases/tag/2.4.0 now released, CPQ queries SpoolManager when attempting to start a new print and will pause with a status message if the printer won't have enough filament to run it.

closed 2023-03-08 by smartin015