smartin015/continuousprint #186: "Waiting for timelapse to render" although it's not activated

TylonHH, 2023-01-08:
I noticed with newest version, that it tooks 30s after finishing the print to clear the bed.

Here is the octoprint.log
```2023-01-08 14:31:36,290 - werkzeug - INFO - 127.0.0.1 - - [08/Jan/2023 14:31:36] "GET /?action=snapshot HTTP/1.1" 200 -
2023-01-08 14:31:46,491 - werkzeug - INFO - 127.0.0.1 - - [08/Jan/2023 14:31:46] "GET /?action=snapshot HTTP/1.1" 200 -
2023-01-08 14:31:57,370 - werkzeug - INFO - 127.0.0.1 - - [08/Jan/2023 14:31:57] "GET /?action=snapshot HTTP/1.1" 200 -
2023-01-08 14:32:08,247 - werkzeug - INFO - 127.0.0.1 - - [08/Jan/2023 14:32:08] "GET /?action=snapshot HTTP/1.1" 200 -
2023-01-08 14:32:18,797 - octoprint.util.comm - INFO - Finished in 2401.816 s.
2023-01-08 14:32:18,798 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2023-01-08 14:32:18,839 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: E_V3_newest_BOTTOM_vase_mode.aw.gcode, owner: None
2023-01-08 14:32:18,979 - werkzeug - INFO - 127.0.0.1 - - [08/Jan/2023 14:32:18] "GET /?action=snapshot HTTP/1.1" 200 -
2023-01-08 14:32:19,310 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2023-01-08 14:32:19,332 - octoprint.filemanager.storage - WARNING - Invalid print time value found in print history for E_V3_newest_BOTTOM_vase_mode.aw.gcode in /home/pi/.octoprint/uploads/.metadata.json: None
2023-01-08 14:32:26,084 - octoprint.plugins.continuousprint - DEBUG - TICK, IDLE, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[], bed_temp=40.02, timelapse_ts=1672822812.5058837
2023-01-08 14:32:31,106 - octoprint.plugins.continuousprint - DEBUG - TICK, IDLE, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[], bed_temp=40.0, timelapse_ts=1672822812.5058837
2023-01-08 14:32:34,280 - octoprint.plugins.continuousprint - DEBUG - TICK, IDLE, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[], bed_temp=40.02, timelapse_ts=1672822812.5058837
2023-01-08 14:32:34,627 - octoprint.plugins.continuousprint - DEBUG - SUCCESS, IDLE, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[], bed_temp=39.98, timelapse_ts=1673184754.6197248
2023-01-08 14:32:34,628 - octoprint.plugins.continuousprint - INFO - _state_printing -> _state_success
2023-01-08 14:32:34,665 - octoprint.plugins.continuousprint - DEBUG - Refreshing UI state
2023-01-08 14:32:34,686 - octoprint.plugins.tracking - INFO - Sent tracking event print_done, payload: {'origin': 'local', 'file': '42161878a9512eda62ae299fe7f2780edbfb744c', 'elapsed': 2401}
2023-01-08 14:32:34,688 - octoprint.plugins.filamentreload - INFO - PrintDone: Disabling filament sensor.
2023-01-08 14:32:35,386 - octoprint.plugins.continuousprint - DEBUG - TICK, IDLE, path=E_V3_newest_BOTTOM_vase_mode.aw.gcode, materials=[], bed_temp=39.98, timelapse_ts=1673184754.6197248
2023-01-08 14:32:35,387 - octoprint.plugins.continuousprint - INFO - Waiting for timelapse to render
2023-01-08 14:32:35,441 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2023-01-08 14:32:35,558 - octoprint.plugins.continuousprint - DEBUG - Refreshing UI state
2023-01-08 14:32:36,184 - octoprint.plugins.continuousprint - DEBUG - suppressing further debug logs for action=TICK, printer state=IDLE
2023-01-08 14:33:06,504 - octoprint.plugins.continuousprint - DEBUG - Refreshing UI history
2023-01-08 14:33:06,581 - octoprint.plugins.continuousprint - DEBUG - _state_success next item "E_LID_V3_vase_mode_deeper_notch_100-5.aw.gcode" (remaining=1, job remaining=1) --> _start_clearing```

smartin015, 2023-01-09:
Thanks for the report! I'm on vacation right now, will take a closer look when I return next week.

smartin015, 2023-03-08:
Man, I missed this one. Sorry about that. 

Can you update to v2.4.0 and tell me if you still see this problem? Thanks!

Villtorb, 2023-11-08:
I'm on v2.4.0 and this is still happening. Weirdly enough, short prints do not seem to trigger the wait. I've observed that longer prints (>30min) have this problem.

TylonHH, 2023-11-13:
Im on 1.9.3 because I run it vie OctoPi. Everything up to date and I have still this issue.