smartin015/continuousprint #187: plugin conflict with gcode editor?

TylonHH, 2023-01-09:
hey,
when I edit a gcode with this [plugin](https://github.com/ieatacid/OctoPrint-GcodeEditor), I got crazy behavior as I mentioned in this issue: https://github.com/ieatacid/OctoPrint-GcodeEditor/issues/33
Maybe you can take a look?

smartin015, 2023-01-09:
Hey @TylonHH - that definitely looks like a conflict with Continuous Print, but sadly not one I'm able to resolve as it appears the GCODE editor plugin is reading the HTML to get the name of the file to edit. 

CPQ injects HTML in order to display an icon next to `.gjob` files, and it looks like that plugin is reading it as part of the file name.

TylonHH, 2023-01-31:
the other plugin was faster :-D
https://github.com/ieatacid/OctoPrint-GcodeEditor/releases/tag/0.2.13
This fixed it.

closed 2023-01-31 by TylonHH