smartin015/continuousprint #190: Better preprocessor debugging / validation

smartin015, 2023-01-12:
**Is your feature request related to a problem? Please describe.**

>  Trial and error is quite a nuisance because I need to start and stop the queue to test events

There should be some way to smooth the process of preprocessor development so folks don't have to test everything in situ.

**Describe the solution you'd like**

TBD, user feedback needed

**Describe alternatives you've considered**

TBD

Terandium, 2023-01-13:
I'd recommend just having the preprocessor run at the click of a button. Maybe being able to set up the state somehow. as for some tests you'll need the metadata.

A proper validator will be a lot more work I assume so if you can, just make the code run(manually) it'll at least stop the hassle of having to save the config start the queue test it out and go back.

smartin015, 2023-01-15:
Thinking about it a bit, I'm now imagining expandable sections at the bottom of each event, after the list of scripts. Each section would show the combined script output and propagation of values across the preprocessors as each event fires, and you'd be able to configure the initial conditions (name of the gcode file that's loaded, its metadata, the state of the queue, external vars etc.)

Whenever a change is made - scripts/preprocessors are added or changed, or scripts are reordered - the section re-executes the sequence and updates the output. That way you can see not just the output of a single preprocessor, but the interaction between preprocessors in the sequence as well as the resulting gcode.

WDYT?

Terandium, 2023-01-17:
That sounds like an awesome idea if you ask me, it'll allow you to ideally test every scenario you can think of and it'll be so much easier to test.

Just a quick note, I do think running the event in "test mode" (lacking a better term) should also run the provided g-code, otherwise, you won't be able to fully test it. Then it's up to the user to make sure that the g-code isn't going to do something potentially destructive to the machine. That might be obvious but that's my opinion at least haha, let me know what u think,

Terandium, 2023-01-17:
The reference image is looking really promising!
I am in doubt if having to run the g-code is necessary, maybe option 2 makes the most sense.
I don't think the g-code will often contain mistakes as most people will make the g-code in notepad++ and just test it by uploading. (At least that's what I did) So if the preprocessor does what you expect it to do, the g-code _should_ fall right into place

smartin015, 2023-03-08:
This is now live in https://github.com/smartin015/continuousprint/releases/tag/2.4.0

closed 2023-03-08 by smartin015