smartin015/continuousprint #192: Reordering prints in a job, doesn't work

Terandium, 2023-01-14:
I haven't provided a system info bundle, as I don't think it's necessary. However, if you need one, I am sorry in advance 😓
This isn't a major issue and I even forgot to even make a bug report about it, but here goes regardless haha.

**Describe the bug**
If you make a new job, with 2 or more different prints.
If you reorder the prints in the job and save them.
Then when you start the queue it ignores the updated order and falls back to the order in which they were added.

**To Reproduce**
Described above.

smartin015, 2023-01-14:
...huh.

Yeah, that's definitely not the intended behavior. No sysinfo bundle required this time :)

I think I know what's going on - the the query used by the queue driver is probably just reading sets in DB insertion order, and not ordering by the `rank` column (which is how order is determined in the UI). 

Since this is a pretty blatant behavioral error, I'll prioritize a fix for it when I get back next week.

smartin015, 2023-03-08:
This is now live in 2.4.0.

closed 2023-03-08 by smartin015