smartin015/continuousprint #206: Nozzle temp set to gcode bed temp randomly between items

Bfreak, 2023-02-10:
Hi all. I have 2 CR30s running CP, using the exact same GCODE. They run the same 4 items 24/7 and have done for months without complaining.

For some reason, seemingly at random, nozzle temp will set to 40c when starting a new item. Sadly octoeverywhere doesn't recognise the absense of the print as a failure so this can go unnoticed. 

I haven't found a way to replicate, given how this very much seems to be at random. I am unable to offer the gcode due to company policy. 

Thank you all for your hardwork on CP!

[octoprint-systeminfo-20230210200120.zip](https://github.com/smartin015/continuousprint/files/10710797/octoprint-systeminfo-20230210200120.zip)

smartin015, 2023-02-17:
Hey @Bfreak - that's an interesting problem. 

I took a look at your sysinfo bundle and found quite a few of these errors from a `crealitycloud` plugin:

```
2023-02-09 18:54:39,108 - octoprint.plugin - ERROR - Error while calling plugin crealitycloud
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 273, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_crealitycloud/__init__.py", line 51, in on_event
    self._crealitycloud.on_event(event, payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_crealitycloud/crealitycloud.py", line 427, in on_event
    if nozzleTemp2 is not None:
UnboundLocalError: local variable 'nozzleTemp2' referenced before assignment
```

I have no idea if that's related to your issue, but you might try disabling the `crealitycloud` plugin and running a few prints to see if you can still trigger the symptom. Clearing `crealitycloud`'s plugin data may also help.

Bfreak, 2023-02-17:
This appears to have been the problem, thank you @smartin015

closed 2023-02-24 by smartin015