smartin015/continuousprint #209: Add manual bed clearing - check if bed is clear using BedReady plugin every x seconds

Defozo, 2023-02-24:
**Is your feature request related to a problem? Please describe.**
I don't find automatic bed clearing useful in some applications. Eg. sometimes when printing with TPU and brim the parts are stuck hard and they move with the bed. Then, I need to ask someone else to clear the bed as I often don't have physical access to the printer I use. This, however, may not happen immediately.

**Describe the solution you'd like**
I'd like to be able to add gcodes to the queue and make the Continuous Print wait for the bed to be ready before starting each print.

**Describe alternatives you've considered**
This probably could be already achieved using Scripts & Preprocessors, but I'm unsure how.

**Additional context**
There are some API/Helpers available for the BedReady plugin if that's necessary: https://github.com/jneilliii/OctoPrint-BedReady/issues/9

smartin015, 2023-02-24:
Hey @Defozo - I see the value in having a way to automatically start up again when conditions are again good for printing. Could you share what your current clearing script is doing, including the GCODE/preprocessors? I might be able to offer more specific advice.

One general concern I have is that BedReady may think the bed is clear while your on-site user is still clearing the bed - which sounds frustrating at best and a hazard to them at worst.

Here are some ideas:

* If you're comfortable with firmware modifications, you could use [action commands](https://smartin015.github.io/continuousprint/action-commands/) to add a menu option on the printer's display that resumes the queue. Your on-site user would press that to continue the queue, implicitly confirming the bed is fully cleared, no fingers in the gantry etc. 
* Wait for multiple "all clear" notices by BedReady before continuing to print. This could be a python script running independently of OctoPrint/CPQ - periodically query BedReady, then issue either a [`set_active` call to CPQ](https://smartin015.github.io/continuousprint/api/#startstop-managing-the-queue) or a [`/job` resume OctoPrint command](https://docs.octoprint.org/en/master/api/job.html#issue-a-job-command) depending on whether your script deactivates the queue or pauses the clearing script. 
* A hybrid of the above - make a physical button, write a phone app, or whatever you're comfortable with that the on-site user can press to issue the above API requests to indicate they've cleared the bed. Maybe you could combine this with the "wait a few minutes" strategy so that it still starts a new print after ~30 mins if they don't confirm.

Defozo, 2023-02-24:
Thank you for your response and ideas.

Just to add, the on-site user who clears the bed is not very tech-savvy and does not have much experience with using OctoPrint or CP. That's why I am trying to find a way to simplify the process for him.

Currently, the on-site user clears the bed manually whenever they see that the printer has finished printing. We have a system where they move the bed along the Y-axis to the back (since when the printer finishes the print it moves the bed to the front), which lets me know that the bed is cleared by looking at the camera. This is also the position in which I took the snapshot with the BedReady plugin.

I appreciate your ideas. I didn't actually know that I could use these REST API commands to resume the job. However, considering the above, I don't think I need any other interface for the on-site user and I think this functionality would be beneficial for other users of CP too.

I'm using the standard setup that comes after installing CP with the profile for Prusa Mini. This is my bed-clearing script:
```
M17 ;enable steppers
G91 ; Set relative for lift
G0 Z10 ; lift z by 10
G90 ;back to absolute positioning
M190 R25 ; set bed to 25 and wait for cooldown

;clear the bed
;first column
G0 X180 Y180 ;move to back corner
G0 Z1 ;come down to 1MM from bed
G0 Y0 ;wipe forward
G0 Y180 ;wipe aft

;second column
G0 X135 Y180 ;move to back corner
G0 Z1 ;come down to 1MM from bed
G0 Y0 ;wipe forward
G0 Y180 ;wipe aft

;third column
G0 X90 Y180 ;move to back corner
G0 Z1 ;come down to 1MM from bed
G0 Y0 ;wipe forward
G0 Y180 ;wipe aft

;fourth column
G0 X45 Y180 ;move to back corner
G0 Z1 ;come down to 1MM from bed
G0 Y0 ;wipe forward
G0 Y180 ;wipe aft

G28 ; home
```

Finished script:
```
M18 ; disable steppers
M104 T0 S0 ; extruder heater off
M140 S0 ; heated bed heater off
M300 S880 P300 ; beep to show its finished
```

Managed Cooldown:
`; Put script to run before bed cools here`

My preprocessors:
https://github.com/smartin015/continuousprint/blob/2cb0dae0262742ba549729d1db20d92f8b516693/continuousprint/data/preprocessors.yaml

However, as I'd like to clarify, I don't use CP when printing parts that have greater bed adhesion (eg. when using a brim) or when I want to make sure the part comes out in excellent quality. This is the case when I want the manual bed clearing. So, I'm not sure about the usefulness of my current scripts in this regard.

I hope this clarifies things. Let me know if you have any further questions or suggestions.

smartin015, 2023-03-06:
Moving the gantry to signal readiness is a pretty smart solution :)

If you'd rather avoid a custom solution, then it sounds like a direct use of the plugin would be the best option. I'm imagining CPQ polling the [check_bed](https://github.com/jneilliii/OctoPrint-BedReady/blob/master/octoprint_bedready/__init__.py#L132) method in BedReady when the next print is about to start.

It wouldn't be too difficult to add - but I've got several other features in the pipe so it may take some time for this to be implemented.

smartin015, 2023-04-06:
Small update - I've been sending PRs to fix up BedReady in preparation for deeper integration. Details at https://github.com/jneilliii/OctoPrint-BedReady/issues/16