smartin015/continuousprint #213: Unable to join network queue

ericwdoyle, 2023-03-04:
**Describe the bug**
Unable to join any network queue. Running multiple instances of octoprint on windows PC. Auto IP nor self assigned IP allow connection. Same issue for all instances attempting to set up LAN queue.

**To Reproduce**
N/A

**Expected behavior**
N/A

**Screenshots**
[image]

**System info:**
[octoprint-systeminfo-20230304142927.zip](https://github.com/smartin015/continuousprint/files/10889544/octoprint-systeminfo-20230304142927.zip)

**Browser logs:**
[Browser logs.txt](https://github.com/smartin015/continuousprint/files/10889545/Browser.logs.txt)

**Additional context**
Add any other context about the problem here.

ericwdoyle, 2023-03-04:
Additionally, I receive this error after saving queue within continuous print settings.

[image]

smartin015, 2023-03-04:
Hey @ericwdoyle - I haven't dug too deeply, but this looks like a dupe of #154. Sadly there's some challenges with cross-compatibility in networking between linux and windows that I haven't yet found time to fix.

ericwdoyle, 2023-03-05:
Hi @smartin015, thanks for the quick reply! Understandable, I imagine there are not many people running these specific conditions. I have only used the printer queue and it has been awesome, but love the idea of the local queue; and would modify my OctoPrint setup as necessary to get this working (I have a 12 printer farm, and growing, that I'd love to get running off of one queue). If I changed the OS of my farm PC to linux do you believe this issue would be resolved? Or should I ideally put each printer on a Pi? Thanks in advance for any support/suggestions!

smartin015, 2023-03-05:
12 printers does sound like a good candidate for setting up a LAN queue :)

Could you describe your setup a bit more? Do you have one PC running 12 printers currently? I've heard some folks having success running containerized versions of OctoPrint to connect to each printer, but I'd be a bit worried about USB bandwidth if you're trying to feed gcode to 12 printers simultaneously from one device.

The "standard" experience as I imagined it would be a raspberry pi per 3d printer, each running octoprint with CPQ installed. That's the most future proof and could allow for other integrations/improvements later down the road (Obico or BedReady for failure detection, SpoolManager for filament management etc.) That's a tad pricy nowadays, though, so if you went that route I'd suggest trying out LAN queues with 2-3 printers first and see how you like the behavior of the queue before shelling out on all 12 :)

I've been working for the last couple months on an overhaul of the existing network printing code - I opened https://github.com/smartin015/peerprint/issues/13 to collect ideas on whether we can cut down installation cost by an order of magnitude or two. That'll be a later-this-year project at best, or a next-year project more realistically.

ericwdoyle, 2023-03-05:
So I currently have 2 printers each running on their own raspberry pis, which have been using CPQ on individual queues.

I have an additional 10 printers that were always unconnected to the network. Within the past few weeks I have set up a windows PC to run multiple instances of octoprint containerized. Currently it is running 5 printers, with plans to run the rest.  UBS bandwidth is a limitation I definitely overlooked, not noticing it yet, but I'd imagine it would be noticeable soon. Maybe that's why I'm having trouble getting more than 2 webcams to stream at a time?

I'm running into a few issues such as difficulty connecting more than a few webcams, limited streaming (.1fsp) to Obico due to windows compatibility problems, and problems with LAN queue. I originally decided to go with the containerized octoprint on a pc rout due to the high cost of pis, the anticipated messiness of running so many pis on my network, speed of gcode upload, and the ease of scaling up, as its basically as easy as copying and renaming the container.

It seems like this was theoretically a good idea lol, yet I overlooked a lot of the software and compatibility issues as I am pretty inexperienced in the software/programming world.

I think your advise sounds right, I'll experiment with running my 2 pi printers on a LAN queue. If all runs smoothly ill probably bite the bullet and 10 more pis. 

That would be awesome to have some type controller pulling code from the network and pushing it to the printer. I will definitely follow that, excited to see what you come up with!

Over the years of 3D printing and operating my farm I have come up with tons of ideas to improve my processes, however most of them are just ideas as I mentioned my programming experience is near to none. I'd like to keep up discussion with you as you continue to develop this program. I been meaning to get more involved in the 3d print community, I've taken in so much from everyone sharing, yet haven't given much back!

Thanks for all your support and your work on the CPQ program, its been really helpful even with me underutilizing its full potential. Will be incredible when I get the whole farm pulling from a single queue!

smartin015, 2023-03-06:
> ...I'd imagine it would be noticeable soon. Maybe that's why I'm having trouble getting more than 2 webcams to stream at a time?

Almost certainly, yeah - webcam bandwidth can be relatively large, especially if you're using an older USB protocol (either by the port/bus or because that's what the webcam supports). 

If you still want to go the Docker route for simplicity, you could look into buying additional PCIe cards with dedicated channels/USB hosts (looking around, I found [this one](https://www.amazon.com/StarTech-com-Express-SuperSpeed-Dedicated-Channels/dp/B00HJZEA2S) as an example). I haven't tried it before, but having each port with its own dedicated channel would potentially eliminate contention between webcams/printers. But not knowing anything about the PC itself, it's possible it wouldn't be enough to handle all the data regardless.
 
> Over the years of 3D printing and operating my farm I have come up with tons of ideas to improve my processes, however most of them are just ideas as I mentioned my programming experience is near to [...]

I do like ideas! If you think of anything that could be improved in CPQ or for print farms in general, definitely open a feature request and I'll see whether/how to fit it in.

> Thanks for all your support and your work on the CPQ program, its been really helpful even with me underutilizing its full potential. Will be incredible when I get the whole farm pulling from a single queue!

You're quite welcome! I'm glad you're getting good use out of it. Onwards to singular queues!