smartin015/continuousprint #220: SpoolManager integration broken

Terandium, 2023-03-08:
**Describe the bug**
I have a g-code script that calibrates the bed, which only runs occasionally.
It extrudes no filament and thus has no filament usage data, the 2.4.0 release freaks out and keeps restarting the print.

I currently have no additional data, as I disabled the spool manager plugin to first finish some prints.
I will in a bit update this issue with proper logs and screenshots

smartin015, 2023-03-08:
Thanks for the report... and sorry the new release broke your setup. Will await the sysinfo and screenshots.

Terandium, 2023-03-08:
No worries, bugs happen!
I have seen the new features in the preprocessors they look dope! Haven't had to use them just yet. But I most certainly will soon enough.

Hope the info provided can help you out

smartin015, 2023-03-08:
Oh I see, so this is a custom (in-queue?) gcode script that doesn't extrude anything. Does SpoolManager complain at all about the print file if you run it outside of CPQ, with the Print button? 

Looking at the plugin code, it seems like it'll [throw a warning](https://github.com/OllisGit/OctoPrint-SpoolManager/blob/master/octoprint_SpoolManager/static/js/SpoolManager.js#L805) but continue anyways if it thinks metadata is missing for the file. I'm torn between emulating SpoolManager's behavior here and preventing the queue from overrunning the spool if a file happens to have a metadata failure. What do you think?

Terandium, 2023-03-08:
If I run the file directly, it will give me a warning about the file containing no metadata.
However I always just click on continue as I know it has no metadata as it extrudes nothing ;)

I agree that this might require a patch from SpoolManager however I am uncertain if that dev replies. I think I have made an issue before there and it got ignored, but I am not 100% certain.

I do think that outside of what SpoolManager does the way CPQ handles it isn't "correct" either, as it just continues retries.

Also last time we spoke I wanted to use SpoolManager as a filament runout sensor, but I came to the conclusion it's too unreliable. I've gotten a filament runout sensor recently and I am using that instead.

Terandium, 2023-03-09:
I have created an issue at the SpoolManager repo, feel free to add any more info there.
https://github.com/OllisGit/OctoPrint-SpoolManager/issues/318

smartin015, 2023-03-09:
> I do think that outside of what SpoolManager does the way CPQ handles it isn't "correct" either, as it just continues retries.

Yeah, that's definitely not great behavior either. I'll update it to pause the queue rather than keep retrying when SpoolManager errors or claims insufficient filament 
 
> Also last time we spoke I wanted to use SpoolManager as a filament runout sensor, but I came to the conclusion it's too unreliable. I've gotten a filament runout sensor recently and I am using that instead.

Interesting - is that due to SM missing extrusion so air-printing still happens? Or was there a different issue?

Terandium, 2023-03-09:
> Yeah, that's definitely not great behavior either. I'll update it to pause the queue rather than keep retrying when SpoolManager errors or claims insufficient filament
That works ^^ I am currently just manually running the calibration script and that works fine. As I only run it when I do some manual intervention anyways haha
> Interesting - is that due to SM missing extrusion so air-printing still happens? Or was there a different issue?
I had some under-extrusion issues, that have been fixed by now.
But the filament isn't 100% accurate if the printer for example does something wrong and it skips the filament.
It loses accuracy, I've had it now once or twice where something went wrong in the middle of the night and it'll keep on printing in the air, as the filament got snapped in half.

So all in all, it's just not an accurate way of measuring it. If everything were to go right all the time, it would. But it simply doesn't always go right so it ends up being wrong. And a DIY filament sensor is like a few euros XD or even free if you have a bl-touch installed. (you will have a z-stop laying around)

Terandium, 2023-03-10:
Heyy last night it had filament left on the spool but the filament manager said it was empty, thus having it spam repeat.
I had to reset my print job history DB, as it had thousands of new entries (I'll remove all duplicates later by SQL, the interface is too slow to do so haha)

Anyways my question is, could you maybe add an option in the settings that will make it so continuous print ignores the values from the spool manager, the integration is neat but it doesn't always work (on the spool managers side) so maybe a potential opt-out, can be good? _(Do you want me to make a separate issue for this?)_

smartin015, 2023-03-10:
Damn. Yeah, probably should add a bulk history deletion option.

Rather than a setting, what about if CPQ skips checks if SpoolManager has no spools loaded (i.e. no "tools" with spools assigned)? I'm concerned yet another behavioral setting would get forgotten and folks would get confused as to why CPQ isn't referencing SpoolManager - this would make it so that spoolmanager configuration dictates CPQ's behavior. WDYT?

Regardless, I think we can keep it in this bug for now until plans have settled :)

Terandium, 2023-03-10:
I use another plugin for history, and that one got spammed. I think your history db didn't register it(unsure)

Anyways I get that an option would confuse people, thus my idea was an opt-out. So if you have spool-manager installed it's enabled by default, but you can disable it if needed.

Because in my case, sometimes the spool manager says a spool is out but there are still, for example, 100 grams left. This would then force me to manually change the spool amount remaining. Which is quite a nuisance as it happens quite often surprisingly.

At least that is my idea, as the integration works great, it's just that my spool manager isn't always 100% accurate as to what's actually on my spool. But I don't use it for that(not anymore at least) I use it for price calculation, history, and government inventory paperwork.

smartin015, 2023-03-10:
> I use another plugin for history, and that one got spammed. I think your history db didn't register it(unsure)

Ah, I see. Yeah, it's unavoidable for CPQ event triggered scripts to end up in another history plugin, since they're treated as scripts themselves.

> ...sometimes the spool manager says a spool is out but there are still, for example, 100 grams left. This would then force me to manually change the spool amount remaining. Which is quite a nuisance as it happens quite often surprisingly.

IMO changing the spool amount is about as much work as changing a CPQ setting in terms of clicks/keypresses. The solution I proposed was instead deselecting the spool as a way to "disable" CPQ spoolmanager checks, i.e. clicking the X button here:

[image]

In other words: no spool selected => no filament checking. Turning it back on is as simple as selecting the spool again.

> At least that is my idea, as the integration works great, it's just that my spool manager isn't always 100% accurate as to what's actually on my spool. But I don't use it for that(not anymore at least) I use it for price calculation, history, and government inventory paperwork.

Agreed, even as a reference it's pretty handy. I do wonder what it'd take to make it more reliable, maybe for SpoolManager to support something like [this spool weight sensor](https://www.youmagine.com/designs/filament-holder-and-scales) for more precise estimates.

Terandium, 2023-04-18:
> IMO changing the spool amount is about as much work as changing a CPQ setting in terms of clicks/keypresses. The solution I proposed was instead deselecting the spool as a way to "disable" CPQ spoolmanager checks, i.e. clicking the X button here:
A bit late but I was thinking of not disabling it per spool, but to to just disable the entire pausing of queue when filament is out.
I need the queue selected like that for, price calculation, print history, etc. And it's accurate enough to show filament extruded.

But I would like the pausing of print when spool is out, disabled. As that functionality only backfires for me. As explained in: https://github.com/smartin015/continuousprint/issues/230

smartin015, 2023-04-19:
Gotcha - thanks for being patient on this as I've been a bit sidetracked with some other projects. Thanks for making this a separate FR as well :) let's continue the disable option convo there and leave this for the retry bug.