smartin015/continuousprint #221: API Connection to Add Set Throws Error 500 when initiated with N8N

nickgabbard, 2023-03-09:
**Is your feature request related to a problem? Please describe.**
I have an inventory and sales management system for my farm.  I am attempting to automatically create sets/jobs for orders as they arrive using an automation tool called N8N that can be triggered from Shopify.

I am having trouble deciphering the octoprint.log entries that are generated when trying to add a set via the API using an automation package known as N8N.

When attempting to use the plugin/continuousprint/set/add endpoint, I'm getting a 500 error and I can't figure out what I'm doing wrong.

Relevant section of octoprint.log
2023-03-09 21:56:05,281 - octoprint - ERROR - Exception on /plugin/continuousprint/set/add [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1519, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1517, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1503, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1575, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1598, in decorated_view
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/octoprint/plugins/lib/python3.8/site-packages/continuousprint/api.py", line 88, in cpq_permission_wrapper
    return func(*args, **kwargs)
  File "/octoprint/plugins/lib/python3.8/site-packages/continuousprint/api.py", line 190, in add_set
    data = self._preprocess_set(dict(**flask.request.form))
  File "/octoprint/plugins/lib/python3.8/site-packages/continuousprint/plugin.py", line 222, in _preprocess_set
    data["path"],
KeyError: 'path'

I can use the get endpoint that shows me the state of the queues, so I'm confident it's not an authentication or access issue.

Details of the request I'm trying to send:
[image]

Send Query Parameters: No
Send Headers: No
Send Body: Yes
Body Content Type: JSON 
Specify Body: Using JSON
JSON: {"path":"4off.gcode","sd":false,"count":1,"jobName":"Test Job","job":""}

Notes:  We don't use SD cards, and I have confirmed that 4off.gcode exists in the uploads folder.
[image]

This is the response I'm getting:
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

**Describe the solution you'd like**
I would like to know are there other places to look to see what data is being passed to the API.

**Describe alternatives you've considered**
None yet.

**Additional context**
Is there something boneheaded I'm missing?

smartin015, 2023-03-10:
No worries, thanks for posting! In fact, could you do me a favor and weigh in on #57 when you've landed on a useful setup? I'd love to have a blessed - or at least functional - set of steps for integrating inventory management.

[This](https://smartin015.github.io/continuousprint/api/#add-a-set) and [this](https://github.com/smartin015/continuousprint/blob/master/api_examples/example.py#L19) may be of use to you - it should work with either a JSON or form-style post request, but you might try the latter just to see if it unsticks anything. ([source code](https://github.com/smartin015/continuousprint/blob/master/continuousprint/api.py#L186) for reference)

HTH

nickgabbard, 2023-03-10:
Ok, so passing the data as a form style post did the trick, but straight JSON did not work for some reason.

smartin015, 2023-03-10:
> Ok, so passing the data as a form style post did the trick, but straight JSON did not work for some reason.

Good to know. Let's leave this bug open for now; I'll investigate before closing.