smartin015/continuousprint #230: Filament runout warning instead of pause

Terandium, 2023-04-18:
**Is your feature request related to a problem? Please describe.**
My spools in spool manager don't always add up to what is precisely on the spool, so now I have to up my spool amount so that it doesn't pause it through the software. Which is annoying because now my price calculations are off. I have a filament runout sensor, so pausing it through software makes it really frustrating (especially with the bug where it keeps repeating and thus overfilling my separate print history database. This bug is already reported, so no worries there)

**Describe the solution you'd like**
I have brought this up before and you didn't want to clutter the settings more, which makes sense. So maybe make it an opt-out, so it's enabled by default but is able to be disabled. Otherwise, add it as a hidden option in the octoprint config file, so people who have an explicit reason to disable it. have the option too.

**Describe alternatives you've considered**
Forking the repo and just stripping the functionality myself, but since my last issues were handled super well and you were very accepting. I figured I'd try it like this first.

smartin015, 2023-04-19:
Yeah, I suppose having a runout sensor is a pretty good reason to not want blocked files based on approximate SM counts. Sure, let's go with "on by default, but you can change the behavior". 

It's going to be several weeks before I can start on this sadly - but I don't plan to release any time soon either, so if that timeline doesn't work for you I'd recommend a fork/disable for now until I can catch up :)

Terandium, 2023-04-20:
All good, I can work around it for now.
Goodluck with your other work ^^

Terandium, 2023-05-02:
If you go to spool manager settings and uncheck: "Warn, if filament is not enough for print job"
It doesn't give a warning thus it won't get triggered here either

closed 2023-05-02 by Terandium