smartin015/continuousprint #231: Running Success Print script triggering continuously

AlexandreJubert, 2023-05-02:
**Issue**

After each print on my queue completes, instead of properly pausing the Success script is getting ran multiple times. I get a notification for it every time.

**Expected behavior**
The success script is ran a single time.

**Script**
Here's the only script that is ran on Print Success:

```
M18 ; disable steppers
M104 T0 S0 ; extruder heater off
M140 S0 ; heated bed heater off
@pause ; wait for user input
@BEDREADY ; make sure the bed is clear before resuming
```

**Log files**
[octoprint-systeminfo-20230502102500.zip](https://github.com/smartin015/continuousprint/files/11374262/octoprint-systeminfo-20230502102500.zip)
[octopi.local-1683037671730.log](https://github.com/smartin015/continuousprint/files/11374269/octopi.local-1683037671730.log)

Thank you for your help and let me know if I can provide more information.

Terandium, 2023-05-02:
Are you using spool manager?

AlexandreJubert, 2023-05-02:
Yes I am using spool manager.

Terandium, 2023-05-02:
What is the amount left on your spool?

Terandium, 2023-05-02:
Can you try to set your spool used to 0, once done does it still behave the same?

AlexandreJubert, 2023-05-02:
I run the same pause script on Awaiting Material event

Terandium, 2023-05-02:
To clarify: Set your spool to full again, so it has lots of filament left. (In the spool manager the physical spool can be ignored)

I am asking this as there is a bug that triggers when the spool doesn't contain enough filament to start another print.

AlexandreJubert, 2023-05-02:
I see, I'll try this. One thing to note, the prints I queued each uses only 0.67g of plastic so with 64g left it should have more than enough.

AlexandreJubert, 2023-05-02:
Trying another queue right now I'll report how it behaves on print completion in 10min~.

Terandium, 2023-05-02:
Ah okay, then what I am saying shouldn't be the issue ^^ As it has enough filament left.

AlexandreJubert, 2023-05-02:
Another thing to note here, the awaiting material script as getting spammed in the same way the print success script was spammed.

AlexandreJubert, 2023-05-02:
Alright this time it skipped straight out of the pause and triggered a print right away without pausing or me resuming the print.

I've attached my latest system info.

[octoprint-systeminfo-20230502120736.zip](https://github.com/smartin015/continuousprint/files/11375331/octoprint-systeminfo-20230502120736.zip)

Terandium, 2023-05-02:
As far as I can tell the success print does get triggered and run, maybe it's empty? Would explain it continuing instantly.
Unsure tbh